bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #69: SPI problem when running Klipper on SKR Pro with TMC5160

Spacemarine2018, 2019-11-05:
The manual does not contain any information on how to wire the TMC5160 V1.2 drivers from BTT to the SKR Pro. Please improve the manual!

I simply plugged the drivers into the SKR Pro without any changes in the jumper configuration, so I have no jumpers underneath the driver and 4 jumpers below the driver on the right side. I can run Klipper and drive my steppers and also set microsteps and current. This means that I can successfully write via SPI. The problem is that I can not read via SPI, I only receive 00000 when doing DUMP_TMC STEPPER=stepper_x

Does anybody else have this problem? Is my wiring wrong?

BastR, 2019-11-07:
I'm running TMC 5160 but with marlin without an issues, simply plug them like TMC2130 but I had to cut a pin like in this issue: https://github.com/MarlinFirmware/Marlin/issues/15296

Spacemarine2018, 2019-11-09:
@BastR Thanks, I tried, but It doesn't make a difference. You are also on SKR Pro and TMC5160 V1.2 from BTT?

BastR, 2019-11-09:
@Spacemarine2018 I've got SKR PRO with TMC5160 from Watterott for X & Y, Z & E are TMC2130.

Spacemarine2018, 2019-11-09:
@BastR 

Ok, so my config might need some other tricks. It seems not a lot of people are running this config. I hope someone from BTT will respond.

Baltazart, 2019-11-12:
a link that might interest you,  https://github.com/sky8000/SKR-Pro-TMC5160-Ender-5-Config-Files-Marlin-2

johngianni, 2019-11-12:
@Spacemarine2018 I'm running Klipper with v1.2 5160s with no problems. Attaching my config.
Still working on the LCD, which is not even giving signs of live :) 

`stepper_x]
step_pin: PE9
dir_pin: !PF1
enable_pin: !PF2
step_distance: .0125
endstop_pin: PB10
position_endstop: 0
position_max: 300
homing_speed: 100

[stepper_y]
step_pin: PE11
dir_pin: !PE8
enable_pin: !PD7
step_distance: .0125
endstop_pin: PE12
position_endstop: 0
position_max: 300
homing_speed: 100

[stepper_z]
step_pin: PE13
dir_pin: PC2
enable_pin: !PC0
step_distance: .0025
#endstop_pin: PG8
endstop_pin: probe:z_virtual_endstop
position_min: -5
position_max: 400

[extruder]
step_pin: PE14
dir_pin: PA0
enable_pin: !PC3
#step_distance: .002
step_distance: 0.000236
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: PB1 # Heat0
sensor_pin: PF3 # T0 Header
sensor_type: ATC Semitec 104GT-2
control = pid
pid_kp = 16.310
pid_ki = .701
pid_kd = 63.680
min_temp: -100
max_temp: 300
#max_extrude_only_velocity: 35
max_extrude_only_accel: 80
max_extrude_only_distance: 120.0
#max_extrude_cross_section=1.28
pressure_advance: 0.06`

`########################################
# tmc5160 configuration
########################################

[tmc5160 stepper_x]
cs_pin: PA15
spi_bus: spi3a
#diag1_pin: PB10
microsteps: 16
run_current: 0.800
hold_current: 0.500
stealthchop_threshold: 250
sense_resistor = 0.075

[tmc5160 stepper_y]
cs_pin: PB8
spi_bus: spi3a
#diag1_pin: PE12
microsteps: 16
run_current: 0.950
hold_current: 0.7500
stealthchop_threshold: 300
sense_resistor = 0.075

[tmc5160 stepper_z]
cs_pin: PB9
spi_bus: spi3a
#diag1_pin: PG8
microsteps: 16
run_current: 1.100
hold_current: 0.450
stealthchop_threshold: 30
sense_resistor = 0.075

[tmc5160 extruder]
cs_pin: PB3
spi_bus: spi3a
#diag1_pin: PE15
microsteps: 16
run_current: 0.700
hold_current: 0.400
stealthchop_threshold: 5
sense_resistor = 0.075`

oldman4U, 2020-10-10:
Please help the community and close the ticket once you do not need it anymore. Thank you

Spacemarine2018, 2020-10-10:
Has BTT improved the manual? Then it can be closed.

oldman4U, 2020-10-10:
Not sure you know that you as a part of the community can do so.

Spacemarine2018, 2020-10-10:
I know that, thanks. My issue has been solved, but I'd like to point out, that although BTT makes great products, their documentation is really bad.

closed 2020-10-10 by Spacemarine2018

oldman4U, 2020-10-10:
I agree 100%. But waiting for BTT does not help other users. Giving a bit back by contributing helps. 

Thank you for closing this ticket