bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #70: [No Issue] Alternative Pin Map

EsserPrototyping, 2019-11-07:
This is an alternative graphic for the pins of the SKR PRO 1.1
Maybe bigtreetech want to copy it to the repo.. ?
[image]

AnHardt, 2019-11-07:
The places for the resistors R61, R65, R67, R69, R70, R71 seem to be not populated on all boards.
(the not populated pads are (in the drawings orientation) near the top and left top of the processor)
They should connect the stepper serial RX and RX pines. 
The TX pins PE4, PE2, PE0, PD2, PD0, PD5 do end at the pads for the resistors. They are nor usable for steppers serial connection.

CleoDarius, 2019-11-08:
Just to mention that since many hotbeds have a led to indicating heating is on the reversed labels for HOT BED mean that the LED indicator does not work since the led needs the correct polarity.

EsserPrototyping, 2019-11-08:
@AnHardt thanks for the info! I don´t know much about using uart with drivers. Are the missing resistors leading to a problem, or are the just not needed (because of one wire uart eventually)?
@CleoDarius good point! Didn´t thought about that!

AnHardt, 2019-11-08:
At the current implementations the TX pins are not used - so currently not a problem.
But in the future someone could try to use them - and will wonder why they don't work.
It would be better to not mention them in the drawing, like they don't in the pin-files.

rflulling, 2019-11-08:
We do not all need these pins. But @AnHardt as you say some one might in the future. So I wonder if they CAN be modded by the user and connected as needed. Or if due to enabled features, etc. There is no reason or connecting would be a bad idea. IE this requires a manufacturers revision for full implementation due to other designs?

EsserPrototyping, 2019-11-08:
I added and changed some things.. thanks for the feedback!
[image]

Pagan-Wizard, 2023-05-12:
Is the endstop pin out for the SKR Pro v1.2 the same as the SKR Pro v1.1?? Specifically, I am asking about voltage, ground, and signal. I am upgrading from a Creality board, and know which wires are which, but I cannot find which pins to plug them into my SKR Pro v1.2. I know where the sockets are on the SKR, but I am asking specifically about the pins within the sockets. The Creality board only uses V and G, but the BTT board uses V, G, and S. I don't have wires for the S, and just want to be sure I am plugging the V and G wires onto the correct pins. I know if I accidentally, or unknowingly, invert them, I can short the mainboard out, so I want to be 110% clear as to where to plug the wires in. Thanks in advance. 
[image]