bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #71: TMC2209 - E0 - TMC Connection Error

trendelkamp, 2019-11-24:
Hi Guys, 

i´am not able to get a TMV 2209 v1.2 from BigTree on E0. Already switched drivers between slots so i can assure that the Hardware Driver is working. 
Any Ideas what is the cause for this issue? 

M122 state the following

> Send: M122
Recv: 		X	Y	Z	E
Recv: Address		0	0	0	0
Recv: Enabled		false	false	false	false
Recv: Set current	800	800	800	800
Recv: RMS current	795	795	795	1436
Recv: MAX current	1121	1121	1121	2025
Recv: Run current	25/31	25/31	25/31	25/31
Recv: Hold current	12/31	12/31	12/31	12/31
Recv: CS actual	12/31	12/31	12/31	0/31
Recv: PWM scale	14	14	14	0
Recv: vsense		1=.18	1=.18	1=.18	0=.325
Recv: stealthChop	true	true	true	false
Recv: msteps		16	16	16	256
Recv: tstep		max	max	max	0
Recv: pwm
Recv: threshold	98	98	658	4347
Recv: [mm/s]		100	100	3	30
Recv: OT prewarn	false	false	false	false
Recv: OT prewarn has
Recv: been triggered	false	false	false	false
Recv: off time	0	0	0	0
Recv: blank time	24	24	24	16
Recv: hysteresis
Recv: -end		-1	-1	-1	-3
Recv: -start		1	1	1	1
Recv: Stallguard thrs	8	8	0	0
Recv: DRVSTATUS	X	Y	Z	E
Recv: stst		*	*	*
Recv: olb
Recv: ola
Recv: s2gb
Recv: s2ga
Recv: otpw
Recv: ot
Recv: 157C
Recv: 150C
Recv: 143C
Recv: 120C
Recv: s2vsa
Recv: s2vsb
Recv: Driver registers:
Recv: 		X	0xC0:0C:00:00
Recv: 		Y	0xC0:0C:00:00
Recv: 		Z	0xC0:0C:00:00
Recv: 		E	0x00:00:00:00	 Bad response!
Recv: 
Recv: 
Recv: Testing X connection... OK
Recv: Testing Y connection... OK
Recv: Testing Z connection... OK
Recv: Testing E connection... Error: All LOW
Recv: 

> 

pins_BTT_SKR_PRO_V1_1.h states the following

> /**
>  * Marlin 3D Printer Firmware
>  * Copyright (c) 2019 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
>  *
>  * Based on Sprinter and grbl.
[... 238 lines cut ...]
>         #endif
>         #ifndef RGB_LED_B_PIN
>           #define RGB_LED_B_PIN PG7
>         #endif
>       #elif ENABLED(FYSETC_MINI_12864_2_1)
>         #define NEOPIXEL_PIN    PG3
>       #endif
>     #endif // !FYSETC_MINI_12864
> 
>     #if ENABLED(ULTIPANEL)
>       #define LCD_PINS_D5  PG3
>       #define LCD_PINS_D6  PG6
>       #define LCD_PINS_D7  PG7
>     #endif
> 
>   #endif
> 
>   // Alter timing for graphical display
>   #if HAS_GRAPHICAL_LCD
>     #define BOARD_ST7920_DELAY_1 DELAY_NS(96)
>     #define BOARD_ST7920_DELAY_2 DELAY_NS(48)
>     #define BOARD_ST7920_DELAY_3 DELAY_NS(600)
>   #endif
> 
> #endif // HAS_SPI_LCD

kamahat, 2019-11-24:
can you post the other files in a pastbin

trendelkamp, 2019-11-25:
For sure: 
[platformio.ini](https://pastebin.com/PmVtPdTy)
[Configuration.h](https://pastebin.com/uU05diWe)
[Configuration_adv.h](https://pastebin.com/Fk3scccu)

Also, if i replace the TMC2209 with a LV8729 in E0 everything works fine.

oldman4U, 2019-12-06:
Hi trendelkamp.

This is maybe a stupid question, but have you ever tried to swap the TMC of E0 and E1 to see if the problem moves with driver.

I have a setup with one extruder, but using the same board and TMC. Would it be helpful for you if i provide the config files and/or a compiled firmware.bin?

MikaistPilot, 2019-12-12:
I have the same Problem. I can connect to E0. But as the Printer starts too print i can not connect to E0. XYZ works still fine.

oldman4U, 2019-12-12:
I removed the StallGuard pin for E0 but left it active for the other 3. Have you tried to remove it to see if it solves the problem.

"Can not connect" you mean the E0 stepper motor does not rotate, right? This works on my machine.

MikaistPilot, 2019-12-12:
No the stepper works. Only the Uart connection not. I removed the Stallgard pins. The Uart works when the printer is not printing. As soon as the print starts the connection ist lost for E0 the rest works fine.

kamahat, 2019-12-12:
Dont' you have something connected on the 2nd expansion port ? because PD2 & PD4 are used for the UART of the E0 and PD2 is also connected on the pin 4 on the 2nd expansion port

MikaistPilot, 2019-12-12:
Fixed the Problem. Withe the latest Marlin Bugfix everxthing works fine.

oldman4U, 2019-12-20:
@trendelkamp 
Have you also been able to fix the reported issue using the latest version of Marlin? Something i found in your Config file which is probably wrong:
#define TEMP_SENSOR_BED 5
I do not believe this causes your issue, but to be sure...

oldman4U, 2020-02-12:
Hello @trendelkamp.

Could you please let us know if the problem is solved in the meantime. If so, kannst du das Ticket bitte schließen, danke.

battlezone, 2020-04-24:
I am having the same issue but on the X-axis. Switching drivers did not help either. Marlin 2.0.5.1

oldman4U, 2020-06-08:
@thinkyhead (Please close this ticket due to inactivity)

Thank you

thinkyhead, 2020-06-14:
Only @trendelkamp or @bigtreetech can close this issue on their repository.

oldman4U, 2020-06-15:
@thinkyhead 

Sorry Scott, this makes sense - my fault. Thank you

@bigtreetech Could you please close this unneeded ticket. Thank you

closed 2020-10-10 by trendelkamp