bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #72: CPU some how is dead. Changed. Board is on but no usb or sd read.

AnyM2k, 2019-11-28:
CPU some how is dead. Changed. Board is on but no usb or sd read.
Any help ? What and how write to new cpu ? Red light is on . Blue light SD card on if connected. 
All endstops is on.

Thank you.

holgin, 2019-11-28:
You need to flash the bootloader to the MCU with ST-LINK. Im trying to do it now, I will let you know if I succeed.

AnyM2k, 2019-11-28:
Thank you for quick answer ! 
May be you have link where to buy and how to use(checked) ? 

Thank you very much !

rflulling, 2019-11-29:
do you have the jumper for the USB / 5V power set? Do you have power set right to each of the supply lines? There are 3 supply lines. Plus USB supply if enabled.

Baltazart, 2019-11-29:
If you need a SKR-Pro v1.1 bootloader, there is a dropbox link, bootloader TFT35v3.0 and SKR-Pro v1.1 to load with STM32CubeProgrammer, USB ST LINK v2 on the SWD connector of SKR-PRO .
https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0/issues/20

AnyM2k, 2019-11-30:
Leds for all 3 power lines on. on usb only 2 on 5V and 3.3V .
Thank you for bootloader ! Ordered ST-Link v2 ... waiting.

holgin, 2019-12-01:
I have used this bootlader and STM32 ST-LINK Utility to upload it and it works, the board is alive :)
[fw_stlink.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/files/3908434/fw_stlink.zip)

AnyM2k, 2020-02-04:
Have st-link V2 on table. At last. 
Any idea how to connect it to SKR-Pro-v1.1 ? 
Soldering to legs ?  Any other ways ? 
Thank you.

AnyM2k, 2020-02-04:
Have some one step by step howto ?
Too many pins , to many errors may occur .

Thank you.

GadgetAngel, 2020-02-07:
Can I use the STM32CubeProgrammer, USB ST LINK v2 on the SWD connector of SKR-PRO to burn a bootloader to the SKR PRO board so I can upload firmware to the board through the USB connection?

Nyxtegida, 2020-02-07:
I received mine with problems, at first it connected to serial but didn't respond to any commands through pronterface. Nor did any difference after trying another firmware in the sd card. Later it didn't even show as a serial but did a read with the st-link. I tried flashing the bootloader from stm32cubeProgrammer, it shown download complete, but didn't confirm the write. After that, reading the mcu gives me full FFFF. Great! XD

AnyM2k, 2020-02-08:
It alive. 
Thank you holgin for fw_stlink file. Connect USB ST LINK v2 to SWD connector of SKR-PRO , ignore reset pin. I use STM32 ST-LINK utility. Open fw , write. 
Thank you.

closed 2020-02-08 by AnyM2k