bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #73: Cannot connect printer as USB STORAGE

karabas2011, 2019-12-02:
I setup printer  
got serial USB working
But for reflashing  I need to remove microsd every time.
How to make USB Storage? Like SKR1.3

oldman4U, 2019-12-06:
Hello karabas2011.

As far as i know, this is so by design. Different boards different feature sets.

Hope this helps

AnHardt, 2019-12-06:
To solve this problem it needs two components.
* Marlin access to the onboard SD-card-slot. That's on its way. https://github.com/LinoBarreca/Marlin/pull/7 is able to use the onboard card but is still under heavy construction. When (if) that's ready to use, it will need several weeks (month?) to backport this to mainstream Marlin (Likely V2.1.0) because it, as it is now, requires changes to all the HALs.
* A USB-composite-device for a common serial and mass-storage device on the, here used, official ST-core. This will probably last a bit longer. Up to now **NO** example for any USB-composite-device for this core exists. But it is possible.

LehrChristoph, 2020-02-26:
Is there anything planned from @bigtreetech to support the USB composite device inside Marlin?