bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #74: Cannot connect, installed both virtual com port driver & marlin one.

mittkonto, 2019-12-07:
So, installed the board.

Steppers: 2pcs tmc2008, with uart activated, according manual.
3pcs of tmc2100 config as a4988 and [spreadcycle](https://www.youtube.com/watch?v=XLkIkG9IMd8) at vref 0.7v there about.

My computer does not recognise it and connecting it to a headless linux SBC, I can't even type lsusb. 

Have not managed to compile either a new marlin nor the one from bigthreetech, so it's what was supplied on the microsd-card.

Any ideas to diagnose further?

Thanks!

georgev1456, 2019-12-11:
I have same problem, i tried to move jumpers for USB Host or USB Flash Drive , changed firmware over 20 times with different code setup no luck. The 5v bridge is works, it seems problems in RX/TX line.

UnderXirox, 2020-02-28:
Hello (sorry for my English) here is a post it reminds me of the concern I had too
here is the post

https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/issues/48

oldman4U, 2020-06-08:
Hello mittkonto.

Could you please be so kind and close this inactive ticket.

Thank you