bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #75: Several servomotor and BL-Touch incompatible ?

yopla31, 2019-12-08:
I installed my SKR Pro 1.1 and the probe BL-Touch with "Marlin 2.0 with bug fixes" and TFT3.5 v2.0. It works fine. Note that the probe use the Zmin (PG8) with this commented lines of pins_BIGTREE_SKR_PRO_V1.1.h : 
```
//#ifndef Z_MIN_PROBE_PIN
  //#define Z_MIN_PROBE_PIN  PA2
//#endif
```
the PA2 is unused now !
But now I'm trying to declare a servo to drive the change of hotend : 
```
#define SERVO0_PIN         PA1  // P0 => BL-Touch
#define SERVO1_PIN         PG5  // P1 => Pour piloter le servo de commande de dual tete
```
The probe does not work, the M280 command P0 Sxxx is interpreted by servomotor, even if I set up another pin. I tried several pins without sucess !
Someone would have tested this configuration? What solution if there is one?

yopla31, 2019-12-11:
The solution was found at the firmware level ([Fix multiple servos with STM32 #16151](https://github.com/MarlinFirmware/Marlin/pull/16151)).

closed 2019-12-11 by yopla31

dineshvyas, 2020-02-28:
PF8 wokred but motors stopped working.