bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #76: Stepper Drivers TMC5160 ver1.3 is not working in SPI or URTA Mode on BIGTREETECH-SKR-PRO-V1.1 Board

georgev1456, 2019-12-11:
Bought the BIGTREETECH-SKR-PRO-V1.1 Board w/ TMC5160 ver1.3 Drivers. 
I cant get it to work, i have tired everything , switching jumpers in SPI Mode or URTA, edited Marlin the Configuration.h , Configuration_adv.h, platformio.ini  i get problems when i change CS Pins  in the Marlin \src \pins directory file "pins_BTT_SKR_PRO_V1_1.h"  i cant build firmware the error come  
on.
 
https://youtu.be/yBzPaFKoo_o
 

[image]

[image]

[image]

i used this manual for steppers.
 https://github.com/bigtreetech/BIGTREETECH-TMC5160-V1.0/blob/master/manual/BIGTREETECH%20TMC5160-V1.0%20manual.pdf

[image]

[image]

[image]

 

NEED HELP......................

oldman4U, 2019-12-12:
Hi george.

I believe you have to use "TMC5160_STANDALONE" with the jumpers set like shown especially on the second picture.

Georg

oldman4U, 2019-12-12:
Sorry, you are using SPI mode, so the jumpers should be fine.....

I write this instead of deleting the message above. Because i use 2209 in UART mode successfully, but had a similar issue using A4988, i believed i could help, but i have no experience with SPI and 5160, so sorry for the confusion.

wesk-ms, 2019-12-17:
Post both of your configuration files and someone might be able to help.

dineshvyas, 2019-12-17:
there are lots of changes to be made. compare my configration_adv.h with yours
[Configuration_adv.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/files/3973672/Configuration_adv.zip)
working configration_adv.h
[image]

dineshvyas, 2019-12-25:
sorry I did't know it was for TMC5160 v1.3
mine are TMC5160 v1.2

georgev1456, 2019-12-26:
so the problem was in the BIGTREETECH-SKR-PRO-V1.1 board , i have received board replacement , now everything works.
[image]
[MARLINE CODE FOR TMC5160.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/files/4001516/MARLINE.CODE.FOR.TMC5160.zip)

closed 2019-12-26 by georgev1456