bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #82: Power selection pins details

frizzle101101, 2019-12-19:
I do not understand why there are 2 jumper pins on the board for the usb host power selection when in the schematics https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/blob/master/Schematic/SKR-PRO-V1.1%EF%BC%88SCH%EF%BC%89.pdf they only describe a 3 pin header so why is there 2 jumpers or why is it a 6 pin header, what are the pins not described.

AnHardt, 2019-12-19:
Are you sure you are looking at the `power-select-jumper` J2 and not on the `USB-slave/host-jumper` J49 ?

frizzle101101, 2019-12-19:
What is the usb disk...why does the board power off usb only with the connection to the usb port and not the usb disk

AnHardt, 2019-12-19:
The 6 pin-jumper J49 has nothing to do with power. It switches the 2 USB-data-lines either to the left USB-connector (right from the green terminals) then acting as a USB-slave (connection to the host), or to the right USB-connector (left from the SD-Card-slot) then acting as a USB-host (for example to read a USB-stick (up to now not implemented in software and probably never will be)).
The 3-pin jumper J2 either connects the internal 5V users to the USB-slave plug (only slaves are allowed to pull power - hosts have to source power), or to the onboard power regulator what makes 5V from the external 12-24V connection.
Open your eyes. Everything is in https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/blob/master/Schematic/SKR-PRO-V1.1%EF%BC%88SCH%EF%BC%89.pdf. You just have to read it.

frizzle101101, 2019-12-19:
Thanks for pointing out the obvious my eyes are open... how about you
create a schamatic that relates the headers to the physical board before
you accuse ppl of not reading. J49 like comon this is terrible documents if
not explained correctly or printed on the board that way

closed 2019-12-19 by frizzle101101