bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #83: USB-Stick support

LehrChristoph, 2019-12-22:
Hi,

I wanted to use the USB-A Port on the board with a USB-Stick, since the internal SD-Card is not usable for printing. Unfortunately there is no documentation neither here nor on the mainline Marlin.

Is there any documentation how to use it?
If not implemented is it planned to add it here or in the Mainline?

Thanks

YaMoef, 2019-12-26:
They are working on it in the marlin software to get an usb recognized as an SD card, but since the usb on the mainboard is connected trough a jumper to the mainboard (D- and D+) but this won't work with the current software only, we are lucky that there is a jumper in between. Because I think maybe you could make a connection overthere to go to another small pcb wich will convert it into some sort of SD card. I saw some boards passing by which could do this an you could hook them up on EXT1 or EXT2 but I'm not surre about this. I'm trying to get some of these board but i can't find them anymore. But in the next version of the board they could better add a MAX3421E which does this some sort of conversion. And the board who are doing this converison I think have this chip. But I think this chip is not an easy to connect and use chip: it might need some configuration and firmware otherwise I would try to create such a board myself.

LehrChristoph, 2019-12-28:
Hi,
Thanks for the info, from the other issue I wasn't sure if this is scope or not.
If I have time I will try to hook up a MAX3421E-board to the printer. At least the description of the USB-Stick -> SD-Card conversion function inside Marlin says it uses a MAX3421E connected to the SPI.

BR Christoph

closed 2019-12-28 by LehrChristoph

YaMoef, 2019-12-28:
@LehrChristoph sorry that I but there is a marlin version which supports onboard sd I look the tread up and post it here

YaMoef, 2019-12-28:
@LehrChristoph https://github.com/MarlinFirmware/Marlin/pull/16260 this is the tread. I came out in this on a tread I made myself en someone replied me yesterday this tread.

LinoBarreca, 2019-12-31:
@LehrChristoph The on board-SD is actually working and faster than LCD-SD. Pull what @YaMoef said and let us know if it works for you (in the pr page)

LehrChristoph, 2020-01-11:
Sorry for the late reply.
I will try it as soon as possible and give you feedback.