bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #85: ESP01 WiFi not enabled

akarnaukh, 2019-12-24:
Specified in the firmware serial port 6. But there is a problem, according to the scheme on the connector pins ch_en, RESET, etc. at start on a pin PG0-CH_EN 0, because of what Wifi Does not start. If you give the command M42 P99 S255, which translates this pin to 1, then everything works.
Question 1 - how to do it automatically at start, what to change in Marlin.
Question 2-why did you bring pins to control the Wifi module from the processor?

UnderXirox, 2020-03-10:
Bonjour
pour active le mode wifi sur ma SKR-PRO

voici a tester pour vous aussi 
crée un fichier  nome  et le placer avec les fichier avec les fichier 

Configuration.h
Configuration_adv.h
and
Configuration_Secure.h

1.  Crée un fichier Configuration_Secure.h
ceci dedans

         /**
          * * To set a default WiFi SSID / Password, create a file called Configuration_Secure.h with
          * * the following defines, customized for your network. This specific file is excluded via
          * * .gitignore to prevent it from accidentally leaking to the public.
          * *
          * *   #define WIFI_SSID "WiFi SSID"
          * *   #define WIFI_PWD  "WiFi Password"
         */
        //#include "Configuration_Secure.h" // External file with WiFi SSID / Password
              #ifndef CONFIGURATION_SECURE_H
                 #define CONFIGURATION_SECURE_H
      
                #define WIFI_SSID "xxxx-xxx-xxx-xxx"  // votre Mots non du reso wifi
                #define WIFI_PWD  "xxxx-xxxx-xxxx-xxx-xxx-xxx-xxx#" // votre mots de passe wifi
          #endif   //Configuration_Secure.h

2. Activer dans le fichier de configuration de Marlin
 Configuration.h

        /**
           * Select the serial port on the board to use for communication with the host.
           * This allows the connection of wireless adapters (for instance) to non-default port pins.
           * Note: The first serial port (-1 or 0) will always be used by the Arduino bootloader.
           *
            * :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
            */
             #define SERIAL_PORT 6  // pour le Wifi car Carte Buguer (1 for TFT uart)

          /**
             * Select a secondary serial port on the board to use for communication with the host.
             * This allows the connection of wireless adapters (for instance) to non-default port pins.
             * Serial port -1 is the USB emulated serial port, if available.
             *
             * :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
           */
               #define SERIAL_PORT_2 1  // -1 for USB 6 pour esp sur la carte mere
             // Je branche les 3 cable EXP1 EXP2 et l'uart de cette facon je peux basculer entre le mode

               #define BAUDRATE 115200

3. Puis dedans le fichier 
Configuration_adv.h

       /**
          * Host Action Commands
          *
          * Define host streamer action commands in compliance with the standard.
          *
          * See https://reprap.org/wiki/G-code#Action_commands
          * Common commands ........ poweroff, pause, paused, resume, resumed, cancel
          * G29_RETRY_AND_RECOVER .. probe_rewipe, probe_failed
          *
          * Some features add reason codes to extend these commands.
          *
          * Host Prompt Support enables Marlin to use the host for user prompts so
         * filament runout and other processes can be managed from the host side.
         */
         #define HOST_ACTION_COMMANDS
              #if ENABLED(HOST_ACTION_COMMANDS)
              //#define HOST_PROMPT_SUPPORT
          #endif

        /**
          * WiFi Support (Espressif ESP32 WiFi)
          */
          // Marlin embedded WiFi managenent
          #define WIFISUPPORT
          // ESP3D Library WiFi management (https://github.com/luc-github/ESP3DLib)
          #define ESP3D_WIFISUPPORT

       #if EITHER(WIFISUPPORT, ESP3D_WIFISUPPORT)
            #define WEBSUPPORT          // Start a webserver (which may include auto-discovery)
            #define OTASUPPORT          // Support over-the-air firmware updates
           //#define WIFI_CUSTOM_COMMAND
          // Accept feature config commands (e.g., WiFi ESP3D)  from  the host

       /**
          * To set a default WiFi SSID / Password, create a file called Configuration_Secure.h with
          * the following defines, customized for your network. This specific file is excluded via
          * .gitignore to prevent it from accidentally leaking to the public.
          *
          *   #define WIFI_SSID "WiFi SSID"
          *   #define WIFI_PWD  "WiFi Password"
          */
          #include "Configuration_Secure.h" // External file with WiFi SSID / Password
       #endif

voila.

PS:  Sans Ca pour moi rien ne fonction pour mon esp-1s  ne fonctionne pas

### et pardon pour ne pas avoir répondue en EN

> Question 2-pourquoi avez-vous apporté des broches pour contrôler le module Wifi à partir du processeur?

Surement pour flache l'esp via la carte SD

Bon Courage.

oldman4U, 2020-06-08:
Hi again.

Same for this ticket. Do you need any further help or can you close it?

Thank you

oldman4U, 2020-06-15:
@bigtreetech Could you please close this unneeded ticket. Thank you

Dosi-Innovate, 2020-07-08:
@UnderXirox thank you. IT WORKED!!! only if i can get the @bigtreetech BTT TFT35 to stop freezing randomly