bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #86: Serial 3 Error

akarnaukh, 2019-12-28:
When I want to select serial 3, a link error occurs.
#define SERIAL_PORT 1
#define SERIAL_PORT_2 3

Linking .pio/build/BIGTREE_SKR_PRO/firmware.elf
.pio/build/BIGTREE_SKR_PRO/src/src/Marlin.cpp.o: In function `manage_inactivity(bool)':
/Users/akarnaukh/SKR_PRO/Marlin/Marlin/src/Marlin.cpp:483: undefined reference to `Serial3'
.pio/build/BIGTREE_SKR_PRO/src/src/Marlin.cpp.o: In function `setup':
/Users/akarnaukh/SKR_PRO/Marlin/Marlin/src/Marlin.cpp:1120: undefined reference to `Serial3'
.pio/build/BIGTREE_SKR_PRO/src/src/core/serial.cpp.o: In function `serialprintPGM(char const*)':
/Users/akarnaukh/SKR_PRO/Marlin/Marlin/src/core/serial.cpp:37: undefined reference to `Serial3'
.pio/build/BIGTREE_SKR_PRO/src/src/core/serial.cpp.o: In function `serial_echopair_PGM(char const*, char const*)':
/Users/akarnaukh/SKR_PRO/Marlin/Marlin/src/core/serial.cpp:41: undefined reference to `Serial3'
.pio/build/BIGTREE_SKR_PRO/src/src/core/serial.cpp.o: In function `serial_echopair_PGM(char const*, int)':
/Users/akarnaukh/SKR_PRO/Marlin/Marlin/src/core/serial.cpp:43: undefined reference to `Serial3'
.pio/build/BIGTREE_SKR_PRO/src/src/core/serial.cpp.o:/Users/akarnaukh/SKR_PRO/Marlin/Marlin/src/core/serial.cpp:44: more undefined references to `Serial3' follow

While switched to Serial 6. 
But it's inconvenient, how do I fix it?

oldman4U, 2020-02-12:
Hello akarnaukh.

What kind of hardware would you like to connect to serial 3? As far as i know, the following Serials can be used with the SKR Pro v1.1:

SKR PRO v1.1
Serial port:
-1 is for USB
1 is for TFT
6 is for ESP-01S

Please let me know

akarnaukh, 2020-02-12:
oldman4U 
to connect SonnOff instead of esp.
All the ports you have listed are just serial ports that are no different from each other. Except maybe USB.
There is no difference what to connect to them. Now for example I use the TTF port to connect to the computer.
The Board has a divorced serial3 port, it should work

oldman4U, 2020-02-12:
OK. Sorry, but i can not help in this case.

oldman4U, 2020-06-08:
Hello akarnaukh.

Seems this ticket has not been used for a while, so it would be great if you could close it.

Thank you

closed 2020-06-21 by akarnaukh