bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #87: SKR PRO V1.1. TMC2208 not working

Lucano99, 2019-12-30:
Hello,

I am struggling with following issue. I have uploaded a firmware (zip attached). i can connect ot the machine, but I cannot move with steppers. It says all the time: "echo:busy: processing"

I have 5x TMC2208, Dual Z, and 1x Extruder. Would you be please so kind and tell me, where is the issue? :(

Thank you!
[Marlin-2.0-Anycubic-Chiron-master.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/files/4010785/Marlin-2.0-Anycubic-Chiron-master.zip)

Klinenator, 2019-12-30:
I have the same exact issue with 2130 drivers

MacBraky, 2019-12-31:
you don't have any powercables connected to the motor or bed connecter, only to the power!? The pro has separate powerinputs.

JulianSimpsonWork, 2019-12-31:
I can confirm a working 2208 configuration on my pro board.  If Marlin started working with my hotend fan properly then I would have everything working I've tried with so far.  You have soldered the pins on the underside of the 2208 haven't you?  I'd start with MacBraky's suggestion as I've always had power to all three supplies from my initial wiring.

oldman4U, 2020-01-15:
Hi Lucano99.

Have you been able to solve the problem in the meantime?

I got my second Pro yesterday and i have A4988, 2208 and 2209 from BTT available for a test. The 2208 and 2209 worked fine on my test system, but i agree, you will have to connect power to the board first.

On your picture i see two "strange" things. First, you have a stepper motor for E1 but no motor connected and second, you have a white cable connected to the USB Host. What is it good for?

kurtisane, 2020-01-18:
Lucano could you please upload your marlin project folder with firmware.bin. I'm facing issues connecting to my board. Maybe your firmware.bin could help me figure out wether or not my board is bricked.

Skippy1488, 2020-01-20:
Looking at your config.h - TMC2280 is what you have defined. Get that inline and it may work.

oldman4U, 2020-02-12:
Seems @Lucano99 does not respond to this ticket. Would it be possible for an admin to close this ticket, please.

oldman4U, 2020-06-08:
@thinkyhead (Please close this ticket due to inactivity)

Thank you

Skippy1488, 2020-06-08:
If I could close it I probably wood.

oldman4U, 2020-06-09:
Well Skippy1488, even you are a good person - for sure, you are not Lucano99 and he/she can close it as you know;-)

I would also close it if i could.....

thinkyhead, 2020-06-14:
Only @Lucano99 or @bigtreetech can close this issue.