bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #88: collect2: error: ld returned 1 exit status *** [.pio/build/BIGTREE_SKR_PRO/firmware.elf] Error 1

John2vdw, 2020-01-01:
Hi, 
my SKR pro v1.1 have been working on this code and a week later without changing the code, doing a rebuild before making changes this error came up.
collect2: error: ld returned 1 exit status
*** [.pio/build/BIGTREE_SKR_PRO/firmware.elf] Error 1

I have saved multiple versions of my settings that worked. Non Compile.
My bigtreetech tft35 does compile.
I used Atom and Visualstudio for compiling, same error on both. 
I tried to download a new copy of marlin and still not compiling.
Bigtree_skr_Pro FAILED

alma2122, 2020-01-03:
Hello, 
got my board today. I have the same problem, also tested with Atom and Visual Studios...

Ausildis, 2020-01-04:
I found a fix:
According to [this](https://github.com/MarlinFirmware/Marlin/commit/04eab3821e6d95fba81970c02b7ba4e7e32bfd9b) I did the following change to the platformio.ini it's somewhere around the line 650

> [env:BIGTREE_SKR_PRO]
> platform          = ststm32
> board             = BigTree_SKR_Pro
> **platform_packages = framework-arduinoststm32@3.10700.191028**
> build_flags       = ${common.build_flags}
>   -DUSBCON -DUSBD_USE_CDC -DUSBD_VID=0x0483 -DUSB_PRODUCT=\"STM32F407ZG\"
>   -DTARGET_STM32F4 -DSTM32F407_5ZX -DVECT_TAB_OFFSET=0x8000
>   -DHAVE_HWSERIAL6
>   -IMarlin/src/HAL/HAL_STM32
> build_unflags     = -std=gnu++11
> extra_scripts     = pre:buildroot/share/PlatformIO/scripts/generic_create_variant.py
> lib_deps          =
>   U8glib-HAL=https://github.com/MarlinFirmware/U8glib-HAL/archive/bugfix.zip
>   LiquidCrystal
>  TMCStepper@>=0.5.2,<1.0.0
>   Adafruit NeoPixel
>   LiquidTWI2=https://github.com/lincomatic/LiquidTWI2/archive/master.zip
>   Arduino-L6470=https://github.com/ameyer/Arduino-L6470/archive/dev.zip
> lib_ignore        = SoftwareSerial, SoftwareSerialM
> src_filter        = ${common.default_src_filter} +<src/HAL/HAL_STM32>
> monitor_speed     = 250000

alma2122, 2020-01-05:
This is the report i get. I have no problems with other boards like SKR 1.3 ...

Processing BIGTREE_SKR_PRO (platform: ststm32@>=5.7.0; framework: arduino; board: BigTree_SKR_Pro)
------------------------------------------------------------------------------------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/BigTree_SKR_Pro.html
PLATFORM: ST STM32 6.0.0 > STM32F407ZG (192k RAM. 1024k Flash)
HARDWARE: STM32F407ZGT6 168MHz, 192KB RAM, 1MB Flash
DEBUG: Current (jlink) External (jlink, stlink)
PACKAGES: toolchain-gccarmnoneeabi 1.70201.0 (7.2.1), framework-arduinoststm32 4.10700.200103 (1.7.0)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 13 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <U8glib-HAL> 0.4.1
|   |-- <Wire> 1.0
|   |-- <SPI> 1.0
|-- <LiquidCrystal> 1.5.0
|   |-- <Wire> 1.0
|-- <TMCStepper> 0.6.1
|   |-- <SPI> 1.0
|-- <Adafruit NeoPixel> 1.3.4
|-- <LiquidTWI2>
|   |-- <Wire> 1.0
|-- <Arduino-L6470> 0.7.0
|-- <SPI> 1.0
|-- <Servo> 1.1.2
|-- <EEPROM> 2.0.1
|-- <IWatchdog> 1.0.0
|-- <Wire> 1.0
Building in release mode
Linking .pio\build\BIGTREE_SKR_PRO\firmware.elf
c:/users/amann/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld.exe: cannot open linker script file stm32f407xg.ld: No such file or directory
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\BIGTREE_SKR_PRO\firmware.elf] Error 1
=============================================================== [FAILED] Took 14.75 seconds ===============================================================

Environment                     Status    Duration
------------------------------  --------  ------------
megaatmega2560                  IGNORED
megaatmega1280                  IGNORED
at90usb1286_cdc                 IGNORED
at90usb1286_dfu                 IGNORED
DUE                             IGNORED
DUE_USB                         IGNORED
DUE_debug                       IGNORED
LPC1768                         IGNORED
LPC1769                         IGNORED
sanguino_atmega644p             IGNORED
sanguino_atmega1284p            IGNORED
melzi                           IGNORED
melzi_optiboot                  IGNORED
rambo                           IGNORED
STM32F103RE                     IGNORED
STM32F103RC_fysetc              IGNORED
STM32F103RC_bigtree             IGNORED
STM32F103RC_bigtree_NOUSB       IGNORED
STM32F103RC_bigtree_512K        IGNORED
STM32F103RC_bigtree_512K_NOUSB  IGNORED
STM32F103RE_bigtree             IGNORED
STM32F103RE_bigtree_NOUSB       IGNORED
STM32F4                         IGNORED
STM32F7                         IGNORED
ARMED                           IGNORED
STM32F103VE_longer              IGNORED
mks_robin                       IGNORED
mks_robin_lite                  IGNORED
mks_robin_mini                  IGNORED
mks_robin_nano                  IGNORED
jgaurora_a5s_a1                 IGNORED
STM32F407VE_black               IGNORED
BIGTREE_SKR_PRO                 FAILED    00:00:14.755
BIGTREE_BTT002                  IGNORED
teensy31                        IGNORED
STM32F103CB_malyan              IGNORED
chitu_f103                      IGNORED
teensy35                        IGNORED
esp32                           IGNORED
fysetc_f6_13                    IGNORED
linux_native                    IGNORED
SAMD51_grandcentral_m4          IGNORED
include_tree                    IGNORED
========================================================== 1 failed, 0 succeeded in 00:00:14.755 ========================================================== 
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.

Ausildis, 2020-01-05:
> PACKAGES: toolchain-gccarmnoneeabi 1.70201.0 (7.2.1), **framework-arduinoststm32 4.10700.200103 (1.7.0)**

@alma2122, you are using a newer framework package that causes these problems change the highlighted line in my last comment and the error will disappear.

schpuntik, 2020-01-08:
diff --git a/firmware/Marlin-bugfix-2.0.x SKR ProV1.1-Dmeo/platformio.ini b/firmware/Marlin-bugfix-2.0.x SKR ProV1.1-Dmeo/platformio.ini
index fba4189..2caf811 100644
--- a/firmware/Marlin-bugfix-2.0.x SKR ProV1.1-Dmeo/platformio.ini      
+++ b/firmware/Marlin-bugfix-2.0.x SKR ProV1.1-Dmeo/platformio.ini      
I did it after the changes
@@ -557,7 +557,7 @@ monitor_speed     = 250000
 [env:BIGTREE_SKR_PRO]
 platform          = ststm32@>=5.7.0
 framework         = arduino
-platform_packages = framework-arduinoststm32@>=3.10700.191028
+platform_packages = framework-arduinoststm32@3.10700.191028
 board             = BigTree_SKR_Pro
 extra_scripts     = pre:buildroot/share/PlatformIO/scripts/generic_create_variant.py
 build_flags       = ${common.build_flags}

CadAxeDesign, 2020-01-13:
Me too but no luck so far. I did try amending the highlighted line and it still the same error.

kongkong2004, 2020-01-15:
you need do change in file: buildroot\share\PlatformIO\boards\BigTree_SKR_Pro.json
replace "stm32f407xg.ld" with "ldscript.ld"

Tinchus2009, 2020-03-11:
I have the same error, but none of the propose solutions worked, any other ideas?

oldman4U, 2020-06-08:
Hello John2vdw.

Seems this ticket has not been used for a while, so could you please close it.

Thank you

oldman4U, 2020-06-15:
@bigtreetech Could you please close this unneeded ticket. Thank you

closed 2020-06-15 by John2vdw

roberth1967, 2020-12-07:
i used on my e3 mini v2.0 the platformio set to STM32F103RC_btt_512K on default_envs