bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #89: PWM Breakout?

mrarmyant, 2020-01-19:
I need 5v PWM control for my laser.  I am guessing one of the extension pins might be pwm, the problem is I don't see how the pinout correlates to anything.  What external pins can be used for pwm and how are they referenced in the config?

BastR, 2020-01-30:
According to the documentation PC8 / PC9 / PE5 / PE6 are PWM capable but PC8 / PE5 / PE6 are already used for the 3 fans AND are 12V. You still have PC9 which is 5V AND on the extension port 1.

[image]

[image]

Otherwise, maybe you can use PA1 or PA2

AnHardt, 2020-01-30:
All that pins have 3.3V - not 5V. There are no 5V pins on this board. You could use all these pins with a level shifter to 5V, or the 12/24V MOSFET pins with a voltage divider.

frank-pian, 2020-06-15:
> 
> 
> All that pins have 3.3V - not 5V. There are no 5V pins on this board. You could use all these pins with a level shifter to 5V, or the 12/24V MOSFET pins with a voltage divider.

HI
Spindle/laser frequency only on supported HALs: AVR and LPC. What should I change if I want to change the frequency in SKR PRO?
thank you very much