bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #91: Can't find #define TEMP_CHAMBER_PIN

ivanlimwk, 2020-01-24:
Want to use chamber thermistor. I am not sure where to place the pins. i have assigned
#define TEMP_SENSOR_CHAMBER 1

I can't find TEMP_CHAMBER_PIN

BastR, 2020-01-30:
[image]

I think you can use T2 or T3 connector to wire your thermistor.

Then you need to specify what kind of thermistor you're using, like `#define TEMP_SENSOR_CHAMBER 1` if you're using _100k thermistor - best choice for EPCOS 100k (4.7k pullup)_, if not just look just above in configuration.h what type of thermistor you have.

And finally I think you can assigned the TEMP_CHAMBER_PIN in your pin files or directly in the configuration.h like that:

#define TEMP_CHAMBER_PIN PF5 //If you have connected the thermistor to T2
or
#define TEMP_CHAMBER_PIN PF6 //If you have connected the thermistor to T3

Plz give me feedback if it works

ivanlimwk, 2020-02-11:
Thank you for the help. 
Sorry for the late respond. I have been busy lately. I'll give your instruction a go.

MrAaXe, 2020-07-20:
I connect sensor to T2 and add "#define TEMP_CHAMBER_PIN PF5" to Configuration.h.
It works,..
See the picture blew.
It shows chamber temperature as 27..
[image]