bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #92: New SKR 1.1 Pro and BigtreetechTFT35-V.2 problem no printer attached

Shamoon78, 2020-01-24:
The board have 4 x TMC5610 drivers attached to it, the main board is communicating to computer normally all working fine(Stealth chop, end stops, heaters, sensors, I tried troubleshooting the connection issue, by using Bigtreetech latest firmware, on both the TFT display and the SKR board, with no success,I compyl Marlin official bug fix 2 myself with all options on serial ports , 1 for TFT, -1 for USB and all other numbers again evrything worked fine on computer but , no hope with the display connection , try to change the baudrate on both the screen and main board marlin to 250000 or 115200 to match but no success , tried to remove all motor drivers and connection except TFT and no success , tried even all types of displays in Marlin define one at a time nothing worked everything working fine on computer but the screen fail to attach to the board.
Please any suggestion, I am happy to order new Display or board but not sure which one have the problem.   (Sorry for my English ,I know it's not good but still ok to communicate not like my TFT display that failing to communicate).

knockoutperformance, 2020-01-25:
I'm having same issue. Trying to use the 3.5 tft v3, wifi, and sd. Have the 2130's working fine. Sd works and wifi, just tft connect to printer error.

Evovita, 2020-01-25:
I had the "no printer attached" problem on and off. Sometimes it was solved by turning my machine (a custom CoreXY using TMC 2130) off then on again. But I also noticed that recompiling the motherboard firmware and reflashing it to the board would make a difference - I have no idea why. With the configuration below however, the "no printer attached" message now only briefly appears on start up, even after multiple on/off cycles. Note that my printer is called 'EV-2' so you will see mention of that in the files:

[Configuration.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/files/4112419/Configuration.zip)

Shamoon78, 2020-01-25:
@ Evovita, thank you for sharing your files and tyring to help, I tried your configuration with no changes  but the issue still there, the only difference now I have a green led light on close to usb connection ,I think is the Wifi module , that iam not using, not sure though, Is there anyway to test the if the serial connection on mainboard or on screen is defective ?(the serial port I mean the 5 pins only connection labled TFT).

Evovita, 2020-01-25:
@Shamoon78 The green led light close to the usb connection shouldn't be on all the time - at least when the printer is idle. When you flash the firmware on the SKR Pro (via the micro SD card), it will flash a few times then stop. This suggests to me that the firmware might be corrupt and if so, explain why, the TFT won't connect to the board. Another issue could be that the micro SD card isn't working - I had this happen to me, and in the end I had to replace the SD card.

The attached file shows what lights should be on when the SKR Pro is idle with the micro SD card inserted, 12/24V (not USB) power mode set via jumper, and USB flash drive mode set via jumper as per the [SKR Pro V1.1 User manual](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/blob/master/manual/SKR%20PRO%20V1.1%20user%20manual%20v1.1.pdf). **1 blue LED for the SD card, 3 orange lights for the board**.

[image]
 
Also, make sure the TFT ribbon cable is correctly connected. At the TFT end it should be connected to the RS232 plug with the single lead connected to the 5V pin (highlighted below, note orientation of the board) - the other pins are combined together on the ribbon cable.

[image]

You might also want to try connecting to the SKR Pro via _Pronterface_, eliminating the TFT from the equation until such time as you know the board is ok.

Shamoon78, 2020-01-25:
Its not the one that flash close to mcu , its the one close to wifi socket

Shamoon78, 2020-01-25:
I tried 3 sd card all flashing and reading from computer ok none have issue

Shamoon78, 2020-01-25:
And exactly that what i have like your attached photo but the green light
come on one time only and stay on until i reset the board and only the blue
and 3 red lights are on .

Shamoon78, 2020-01-25:
@Evovita , iam not sure if i attached the ribbon cable wrong the first time , that might damage the board hardware ? Or only can corrupt bootloader of mcu , and even if its damaged iam not sure what to order new tft or board iam worry to plug any of them on new ones.😕

Evovita, 2020-01-26:
@Shamoon78 As I said earlier, try connecting to the board using _Pronterface_ (unplug the TFT altogether) - if it works e.g. you can move your axis, heat up bed/extruder etc then you at least know the SKR Pro is unlikely to be the culprit.

Evovita, 2020-01-26:
And...after a few more tweaks to my Marlin firmware that were limited to acceleration settings and adding some comments, and then reflashing the board...the problem is back. Restart several times, no change. Recompile the firmware then reflash again, "no printer attached" quickly disappears. It is as if the delay in applying the firmware may help in the TFT serial port connection. Either way, the inconsistent results is unacceptable. This is unlikely to be a loose/intermittent cable connection issue as I have my plugs held in place with a dab of hot glue.

rndsum, 2020-01-26:
Make sure your baud rate is set to
#define BAUDRATE 115200

Their TFT's can't handle much faster than that.

Shamoon78, 2020-01-26:
As i mentioned in the main post everything is working fine when controlled
by computer even i can print normally through usb port

Shamoon78, 2020-01-26:
I Tried that as well does not solve the problem

Shamoon78, 2020-01-26:
@ Evovita so you have the issue again but what you mean by "delay in
applying the firmware"?

Evovita, 2020-01-26:
Eureka! The problem - at least for me - turned out to be the a **faulty TFT ribbon cable connector at the motherboard end**. Despite making sure it was plugged in snugly numerous times, it appears the pins were not consistently making a solid connection and would therefore periodically disconnect which coincided with me applying firmware updates etc. My temporary fix until I get a new cable/connector is to tape the cable down so it flexes the connector end and thus makes a good (better) connection. After many restarts, it still works AOK and the "No printer attached" message disappears almost instantly every time.

I've already had 2 x BigTreeTech micro USB cards fail on me so it wouldn't surprise me the quality of the TFT ribbon cables is 'variable'.

oldman4U, 2020-01-30:
As far as i know, it is not possible to use the USB, the TFT AND the WIFI at the same time on the board, because only 2 serials can be defined.

-1 is for USB/CDC
1 is for TFT
6 is for ESP-01S

In any case, i would recommend to remove hardware (WIFI) and make a stripped down firmware until the most important parts are working properly. After that, i would add functions based on priority.

Hope this helps

GiovaneMunhoz, 2020-02-01:
@Shamoon78  The problem for me was wrong number on serial port, this is my configs and it works for me, even with esp-01.

Configuration.h
#define SERIAL_PORT 6
#define SERIAL_PORT_2 1

knockoutperformance, 2020-02-04:
got it working with this and using the fork multispi for marlin bugfix:
Configuration.h
#define SERIAL_PORT 1 // tft
#define SERIAL_PORT_2 6 // esp3d
#define BAUDRATE 250000 // both esp3d and tftv3

oldman4U, 2020-06-08:
Hello Shamoon78.

Seems this ticket has not been used for a while, so could you please close it.

Thank you

oldman4U, 2020-10-10:
Please help the community and close the ticket once you do not need it anymore. Thank you

abdulrehaman72, 2020-11-04:
Hi all, I have an issue while using BIGTREETECH SKR-PRO-1.1. After uploading the MARLIN 2.0 in to the board, I checked with each and every motor orientation using Pronterface.exe by connecting with USB (serial port's) . The motors were working fine. Now I'm trying to connect the board through pronterface, it's giving an issue like no serial port is connected. I checked thoroughly, no short-circuit in board, no wrong connections to the board,checked by connecting it with many laptops. Its showing the same issue like ' SERIAL  PORT NOT FOUND '.BL TOUCH is probing fine and input to the board is fine even TFT is glowing but it's not functioning. I tried resetting and rebooting the board still my issue was not resolved yet. My doubt is whether any internal issue with communication chip. I request you to give me a solution for this issue. Will be waiting for your support. Thank you.