bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #94: SKR 1.1 pro disconnecting and freezes short after turning on

k-rzr, 2020-01-27:
Hello everyone I have some strange issue with my skr 1.1 pro - when I turn on printer everything works fine (sending commands through usb, moving motors etc.) for like 30 seconds to 1 minute, after that it freezes and doesn't respond to anything (touchscreen turns off), neither through usb connection nor using any of the buttons on the printer, lights on the board are on. It happens no matter if it's connected to pc with usb (using pronterface), using only usb power or one form the power supply. I'm using it with tmc 2208, 12v power supply, touchscreen is from BTT TFT24-V1.1, triple-checked all connections. If it matters motors, end-stops, heaters are from anet a6. My board on picture below:

[image]

oldman4U, 2020-01-28:
Hi k-rzr.

Have you tried to disconnect the USB cable to see if this makes a difference?

oldman4U, 2020-01-28:
Hit Comment and the same time another question came up.

Are you using Dual-Z or 2 Extruders?

k-rzr, 2020-01-28:
I'm using dual Z.
I think I tried yesterday running it with disconnected usb cable but I'm not sure. I'll check it out when I get home.

oldman4U, 2020-01-28:
In case you are using dual-z like me, you can remove the last stepper driver because the Pro provides built-in dual-Z. So you have connector X and Y, each with its own dedicated stepper driver . Then you have 2 (Z) connectors sharing the third stepper driver and another stepper driver for the Extruder. In Marlin you configure a single Z stepper and also all the other dual-Z settings and options can be ignored.

USB i asked, because Cura has an issue, causing a TMC Connection error......

k-rzr, 2020-01-28:
Ok so I checked, it happens even with the unplugged USB cable - the printer after turning on works normally for about 30 seconds, then restarts and about 5 seconds later LCD turns off, LEDs on board are still on and you can't do anything. After turning power off and on printer still does not respond, after some time with no power on it returns to the initial state when it works for a half of a minute. I think I'm gonna examine the board if there's no some short circuits, especially near the mounting bolts.
Removing unnecessary driver didn't improve anything.

k-rzr, 2020-01-28:
Seems like I found the cause - it was my self-made step motor direction reverser that consists of a extension cable and angle 4pin socket XD which is strange because it caused no problems with the standard anet a6 mainboard, right now skr board runs for a few minutes without any issues so far so I'm closing issue

[image]

closed 2020-01-28 by k-rzr