bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #95: 35V on motor inputs?

blanosko, 2020-01-29:
Hi, I hope this is the right place to question this. Is it possible to connect say 35V on motor inputs on this board? In manual it says 12V/24V, but I know that it can be just badly written by manufacturer. So is here anyone who tried this and can confirm my assumption that it can?

Thanks

BastR, 2020-01-30:
The only thing that limits the voltage of the motors between the screw connector and the driver pins are the capacitors. They are 35V rated but it's better to leave a safety margin and apply a lower voltage, the information I found on the internet says that you have to leave a 30% margin, that is to say apply a voltage of 24V maximum.

So, you can apply any voltage you want until you leave a sufficient margin below 35V.

blanosko, 2020-01-31:
Really thanks for answer. I think I figured out schematics so I can support your answer.

Here is schematics of main connector on board. There is clearly visible that motor power is connected directly to VMOTO (VM) pin of stepper drivers.
[image]

And here is schematics of actual stepper driver. So if you look on right top side there is VM pin and only one capacitor (in this case capacitor C7) with value 35V/100uF.

[image]

Thats for X driver, so for another drivers value of capacitor is the same (100uF), just name of capacitor is diferent:

X - C7
Y - C10
Z - C16
E0 - C8
E1 - C14
E2 - C17

And these capacitors are located under every stepper driver connectors (they are easily visible)

[image]

So if anyone needed to connect higher voltage then 24V, everything you need to do is desolder the capacitors and solder new ones with higher voltage rating. And you can power motors with whatever voltage you want.

Again, thanks @BastR for his quick reply. Hope this can help to anyone with my problem.

GadgetAngel, 2020-02-01:
@BastR, @blanosko  Have you connected an SSR up to the SKR pro board to power an AC bed?? I am having a hard time seeing how to do an SSR with the Bed power being separate, any suggestions on this subject would be helpful

BastR, 2020-02-01:
@GadgetAngel Yes, I have an SSR connected like that:

[image]

I had to reverse the positive and negative in the SSR input to make it work. It's "normal", I saw an issues that talked about it on that GitHub.

drmulligan, 2020-02-01:
If I read your schematic correctly, you can just desolder pin 16 on your driver, flip it so it sticks up and does not make contact with the SKR board.  Also solder a pin which goes up through the driver to GND on pin 15 so it connects to both the SKR board and up.  Then wire your cap and >24v power supply to a two pin female connector and plug it in to the top of the driver.  Make sure your >24v power supply line to the drivers is fused.

blanosko, 2020-02-03:
> If I read your schematic correctly, you can just desolder pin 16 on your driver, flip it so it sticks up and does not make contact with the SKR board. Also solder a pin which goes up through the [...]

Yeah, you can just do that. But this board specificaly has 3 independent input rails. So it would be great to make them useful and without wires all over the board. 

But yeah, if your board doesnt have this feature (most boards dont) you can do that.

oldman4U, 2020-06-08:
Hello blanosko.

Seems that this ticket is not needed anymore. So could you please close it.

Thank you

closed 2020-06-15 by blanosko