bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #96: Sensorless homing + Skr Pro 1.1 + TMC5160 + Marlin 2.0 = is that possible ?

GiovaneMunhoz, 2020-01-31:
Hello guys, i'm trying to setup a sensorless homing on tmc5160 (spi) and skr pro, using Marlin 2.0, but I did every single sh*** that I found on google, but the endstops dont trigger, Im about to give you, becouse there's is nothing that I didnt do.

But, if someone has figured out how to make it work, I Will be very happy to know how.

Thanks.

GadgetAngel, 2020-02-01:
@GiovaneMunhoz If you setup Sensorless Homing on all your axis, then by definition of Sensorless homing you will not need to trigger your ENDSTOPs. When you do a M119 all your endstops should show OPEN, that’s assuming you defined Sensorless Homing on X, Y and Z. I hope this helps

GiovaneMunhoz, 2020-02-01:
@GadgetAngel of course my endstop will not trigger, couse' Im using sensorless homing! But when I home any axis, the stallguard dont stop the axis when hit the corner, my machine keeps pushing the axis through the corner even when gets stuck. 
So sensorless homing dont work, it dont sense when is on end of course.

Mariu86, 2020-02-01:
You must change the sensitivity for sensorless homing!

GiovaneMunhoz, 2020-02-01:
@Mariu86 I've tried every single sensitivity number possible. From -64 to 64, that is the range that I've found of tmc5160, tell me if im wrong.

GadgetAngel, 2020-02-01:
@GiovaneMunhoz Can you post a link to your Marlin files?? Like put the whole Marlin folder on Google Drive and share it to the public by supplying us with the link to the top Marlin folder. If you do that I could look at your configuration of Marlin and see if something is off.

GadgetAngel, 2020-02-01:
@GiovaneMunhoz Did you connect CLK pin on the TMC5160 to the GND pin on the TMC5160 before pugging them into the SKR PRO V1.1 board?? Here is a file that shows All the Stepper Motor Driver configuration for the SKR PRO board. It was created by @rflulling. Here it is: https://doc-0k-c4-docs.googleusercontent.com/docs/securesc/seim4417k8n16rvc6rl28ib17otc3iek/3ohsos0r5999avqtuin34rmgbvo6bq7t/1580551200000/06483777329211841429/05169398407596378373/1CR7NcZsWs012gB1fWaCSrZSwWh_6bvzq?e=download&nonce=atcujea3apa26&user=05169398407596378373&authuser=0&hash=rsfcp8t4m3890olu52lalaj8va4kdtqd

GiovaneMunhoz, 2020-02-01:
@GadgetAngel I didnt solder CLK pin to GND, on my research I found that is not needed on BTT TMC5160 v1.2, even on v1.2 I still have to do this?

GiovaneMunhoz, 2020-02-01:
@GadgetAngel  Check my marlin files: https://drive.google.com/open?id=1FXjLBxjaVRBTd6YxUVsU0fO5yQepG6gC

GadgetAngel, 2020-02-01:
> @GadgetAngel I didnt solder CLK pin to GND, on my research I found that is not needed on BTT TMC5160 v1.2, even on v1.2 I still have to do this?
> 
> Check this info:
> [image]

Check with. Multi-Meter to see if the CLK is going to ground. I see conflicting info on weather this fix has been implemented. So depending on when you bought your TMC5160, it could or not be fixed. Again CLK pin on the TMC5160 needs to be going to GND. Just check the CLK pin to ensure this is happening.

GiovaneMunhoz, 2020-02-01:
@GadgetAngel Thanks for the tips, I will check. 

What about my marlin configs, is everything ok?

GiovaneMunhoz, 2020-02-01:
> > @GadgetAngel I didnt solder CLK pin to GND, on my research I found that is not needed on BTT TMC5160 v1.2, even on v1.2 I still have to do this?
> > Check this info:
> > [image]
> 
> Check with Multi-Meter to see if the CLK is going to ground. I see conflicting info on weather this fix has been implemented. So depending on when you bought your TMC5160, it could be fixed OR it could NOT be fixed. Again check the CLK pin on the TMC5160 to ensure it is going to GND with your multimeter.

I did What you said using multi-meter, and all my tmc5160 v1.2 have CLK and GND conected. I used the continuity test to check.

GiovaneMunhoz, 2020-02-01:
@GadgetAngel One question, the only wiring I have to do, is solder the diag pin on stepper driver and plug it on the board, right?

GiovaneMunhoz, 2020-02-01:
Thanks everyone who try to help, Reading about stallguard I saw that stallguard Will only work when stealthchop is disabled and in my case mas enabled, know sensorless homing is working. 
Marlin site:
"StallGuard is only active when the driver is in spreadCycle mode."

So I had to disable stealthchop and marlin Will auto switch to spreadcycle and sensorless Will work. Using tmc5160 v1.2 I set sensitivity at 3 or even 2 for X and Y.

Thanks @GadgetAngel for helping.

closed 2020-02-01 by GiovaneMunhoz

GadgetAngel, 2020-02-02:
> @GadgetAngel Thanks for the tips, I will check.
> 
> What about my marlin configs, is everything ok?

I will look at it tomorrow, I have been testing my stepper drivers on the SKR PRO V1.1 board today. I have A4988, DRV8825, LV8729, TMC2100, TMC2130, TMC2208,TMC2209,TMC5160. I only got the A4988 and DRV8825 today. But I will look at your conf files tomorrow. I have a copy of Marlin already done for my TMC5160. I just have not tested it yet.

GadgetAngel, 2020-02-02:
> Thanks everyone who try to help, Reading about stallguard I saw that stallguard Will only work when stealthchop is disabled and in my case mas enabled, know sensorless homing is working.
> Marlin [...]

I was not aware of this either, thanks for letting me know

GiovaneMunhoz, 2020-02-02:
> I will look at it tomorrow, I have been testing my stepper drivers on the SKR PRO V1.1 board today. I have A4988, DRV8825, LV8729, TMC2100, TMC2130, TMC2208,TMC2209,TMC5160. I only got the A4988 and [...]

Dont bother to check those files, I have changed. I Will up again my files with changes that make my print work, maybe help you or someone.

DroneMang, 2020-04-16:
Does anyone on this thread have their current config and configuration advanced files for the SKR pro and TMC5160? I have been working with the 2209's but I can't seem to get above 350 milliamps without strong active cooling and I would like to not have a fan in the case. Thanks!

sn0wrider, 2020-04-21:
> Thanks everyone who try to help, Reading about stallguard I saw that stallguard Will only work when stealthchop is disabled and in my case mas enabled, know sensorless homing is working.
> Marlin [...]

Hi, can you put works configuration files to google drive?
Best rgs

GiovaneMunhoz, 2020-04-21:
> Does anyone on this thread have their current config and configuration advanced files for the SKR pro and TMC5160? I have been working with the 2209's but I can't seem to get above 350 milliamps without strong active cooling and I would like to not have a fan in the case. Thanks!

Im using tmc5160 with 300-400 mah with no cooling at all, they run very cold. Just lower the mah.

GiovaneMunhoz, 2020-04-21:
> Hi, can you put works configuration files to google drive?
> Best rgs

I gave up of using sensorless, they dont work at all. Im using endstop in all axis.

DroneMang, 2020-04-21:
Here are working config files for an SKR Pro V1.1 with TCM 5160 V1.2 on an FT-6, BLTouch 3.1 in 5V and SW mode, sensorless homing. Sensorless homing works fine.  800 mah for XYZ and sensitivity set  M914 X2 Y4.

[Marlin-2.0.x-5160.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/files/4511154/Marlin-2.0.x-5160.zip)

5V mode may not be necessary, but having probe issues per here: https://github.com/MarlinFirmware/Marlin/issues/17337

so I am trying to diagnose.

DaVincIan, 2024-12-23:
Thanks for everyones contributions. Herewith my experience to get sensorless homing ons SKR Pro V1.2 and TMC 5160 drivers.
I have a CoreXY RepRap printer.

I am up to date with Marlin FW: 2.1.2.5
In no particular order:

All these are in Configuration_adv.h

I had to disable the Quickhome
//#define QUICK_HOME                          // If G28 contains XY do a diagonal move first

I set my Stall sensitivity to 4 for X & Y
   #define X_STALL_SENSITIVITY  4

Disable stealthchop. I disable E stealthchop to improve my extruder torque.
// @section tmc/stealthchop
  #if HAS_STEALTHCHOP
//    #define STEALTHCHOP_XY
//    #define STEALTHCHOP_Z
    #define STEALTHCHOP_I
    #define STEALTHCHOP_J
    #define STEALTHCHOP_K
    #define STEALTHCHOP_U
    #define STEALTHCHOP_V
    #define STEALTHCHOP_W
//    #define STEALTHCHOP_E
  #endif