bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #97: Can not compile Marlin 2.0.3 with TMC 2209 for SKR PRO v1.1

Necropaw, 2020-02-01:
As the TMC Stepper driver from https://github.com/bigtreetech/TMCStepper are outdated it is not possible to use the current Marlin version.

### Current behavior
When building with in `platform.ini`:
```
lib_deps =
  LiquidCrystal
  TMCStepper=https://github.com/bigtreetech/TMCStepper
  Adafruit NeoPixel
```

The following error is produced:
```
Compiling .pio\build\BIGTREE_SKR_PRO\lib393\TMCStepper@src-28254114d684293458a93e3358bd6db0\source\TMC5160Stepper.cpp.o
Compiling .pio\build\BIGTREE_SKR_PRO\lib393\TMCStepper@src-28254114d684293458a93e3358bd6db0\source\TMCStepper.cpp.o
Compiling .pio\build\BIGTREE_SKR_PRO\lib108\Adafruit NeoPixel_ID28\Adafruit_NeoPixel.cpp.o
Compiling .pio\build\BIGTREE_SKR_PRO\lib108\Adafruit NeoPixel_ID28\esp8266.c.o
.pio\libdeps\BIGTREE_SKR_PRO\TMCStepper@src-28254114d684293458a93e3358bd6db0\src\source\SW_UART.cpp:81:3: error: 'stimer_t' does not name a type; did you mean 'timer_t'?
   stimer_t SWSerialTimerHandle;
   ^~~~~~~~
   timer_t
.pio\libdeps\BIGTREE_SKR_PRO\TMCStepper@src-28254114d684293458a93e3358bd6db0\src\source\SW_UART.cpp:98:25: error: variable or field 'SWSerial_Handler' declared void
   void SWSerial_Handler(stimer_t *htim) {
                         ^~~~~~~~
.pio\libdeps\BIGTREE_SKR_PRO\TMCStepper@src-28254114d684293458a93e3358bd6db0\src\source\SW_UART.cpp:98:25: error: 'stimer_t' was not declared in this scope
.pio\libdeps\BIGTREE_SKR_PRO\TMCStepper@src-28254114d684293458a93e3358bd6db0\src\source\SW_UART.cpp:98:25: note: suggested alternative: 'timer_t'
   void SWSerial_Handler(stimer_t *htim) {
Compiling .pio\build\BIGTREE_SKR_PRO\libd81\U8glib-HAL_ID1932\U8glib.cpp.o
                         ^~~~~~~~
                         timer_t
.pio\libdeps\BIGTREE_SKR_PRO\TMCStepper@src-28254114d684293458a93e3358bd6db0\src\source\SW_UART.cpp:98:35: error: 'htim' was not declared in this scope
   void SWSerial_Handler(stimer_t *htim) {
                                   ^~~~
Compiling .pio\build\BIGTREE_SKR_PRO\libd81\U8glib-HAL_ID1932\U8glibPrint.cpp.o
.pio\libdeps\BIGTREE_SKR_PRO\TMCStepper@src-28254114d684293458a93e3358bd6db0\src\source\SW_UART.cpp:98:35: note: suggested alternative: 'tm'
   void SWSerial_Handler(stimer_t *htim) {
                                   ^~~~
                                   tm
.pio\libdeps\BIGTREE_SKR_PRO\TMCStepper@src-28254114d684293458a93e3358bd6db0\src\source\SW_UART.cpp: In member function 'void SoftwareSerial::timerInit()':
.pio\libdeps\BIGTREE_SKR_PRO\TMCStepper@src-28254114d684293458a93e3358bd6db0\src\source\SW_UART.cpp:105:5: error: 'SWSerialTimerHandle' was not declared in this scope
     SWSerialTimerHandle.timer = SW_SERIAL_TIMER;
     ^~~~~~~~~~~~~~~~~~~
.pio\libdeps\BIGTREE_SKR_PRO\TMCStepper@src-28254114d684293458a93e3358bd6db0\src\source\SW_UART.cpp:105:5: note: suggested alternative: 'HardwareTimer_Handle'
     SWSerialTimerHandle.timer = SW_SERIAL_TIMER;
     ^~~~~~~~~~~~~~~~~~~
     HardwareTimer_Handle
compilation terminated due to -fmax-errors=5.
*** [.pio\build\BIGTREE_SKR_PRO\lib393\TMCStepper@src-28254114d684293458a93e3358bd6db0\source\SW_UART.cpp.o] Error 1
```

### Expected behavior
It is possible to use BTT TMC Stepper 2209 with Marlin >= 2.0.2

GadgetAngel, 2020-02-01:
@Necropaw Try compiling Marlin 2.0.2 using the default Platformio.ini. I mean do not change the compile directive for the TMC stepper library. BIGTREETECH stepper motor library is out of date.  Use : `lib_deps =
  LiquidCrystal
  TMCStepper@>=0.6.1,<1.0.0
  Adafruit NeoPixel’

Necropaw, 2020-02-01:
@GadgetAngel this worked.

The documentation in https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/blob/master/firmware/Readme.md is outdated.

GadgetAngel, 2020-02-01:
Yes I agree. BIGTREETECH needs to update their online documentation because it is causing confusion with their users

Mariu86, 2020-02-02:
@Necropaw You successfully compiled the 2.0.3 version of Marlin with tmc2209?

GadgetAngel, 2020-02-02:
> @Necropaw You successfully compiled the 2.0.3 version of Marlin with tmc2209?

I think @Necropaw did get it to compile BUT leave the ticket OPEN because others need to be aware that the documentation (https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/blob/master/firmware/Readme.md) is WRONG by telling the user to change the Platformio.ini. 
The user should use the default Platformio.ini file and just change the following item `default_envs = BIGTREE_SKR_PRO`. This ticket should ONLY BE CLOSED after BIGTREETECH changes the documentation at https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/blob/master/firmware/Readme.md

LegendAlexander, 2020-04-09:
I am actually running that build just fine. This is what i used [https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/tree/master/firmware/Marlin-bugfix-2.0.x%20SKR%20ProV1.1-Dmeo/Marlin-2.0.xSKR%20ProV1.1-Demo](url)

You may also find this video helpful:[https://youtu.be/VsKHwYtqLz0](url)

the only thing i can't get is the sd card slot to work for printing. And i don't know what else i could use to print g code files....

GhostlyCrowd, 2020-04-09:
> I am actually running that build just fine. This is what i used [https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/tree/master/firmware/Marlin-bugfix-2.0.x%20SKR%20ProV1.1-Dmeo/Marlin-2.0.xSKR%20ProV1.1-Demo](url)
> 
> You may also find this video helpful:[https://youtu.be/VsKHwYtqLz0](url)
> and this one 2208 and 2209 are very similar: [https://youtu.be/3Z4aAzuzu44](url)
> 
> The only thing i can't get is the sd card slot to work for printing. And i don't know what else i could use to print g code files....

Configuation_adv.h

Change 
#define SDCARD_CONNECTION LCD

to 

#define SDCARD_CONNECTION ONBOARD

LegendAlexander, 2020-04-09:
i did but no luck

GhostlyCrowd, 2020-04-09:
> i did but no luck @GhostlyCrowd

Post your configs and I can compile with the latest pull of marlin if you want 

Maybe it's a bug in the BTT source, I'm waiting for my board to arrive, so I cannot test though.

GhostlyCrowd, 2020-04-09:
Also forgot If are you using one of these in the SKR Pros pins file?

 // Use one of these or SDCard-based Emulation will be used
//#define SRAM_EEPROM_EMULATION   // Use BackSRAM-based EEPROM emulation
#define FLASH_EEPROM_EMULATION  // Use Flash-based EEPROM emulation

PetersonAugusto, 2020-10-12:
Hello friends. Good night, how are you? I hope they are. PLEASE, could you help me with the firmware of my coreXY printer. I think I started in 3D printing with the worst luck possible. I bought a 3D coreXY hyperCube printer project, and I replaced the old board with SKR PRO v1.2 + tmc 2209+ TFT24 and I'm having trouble programming the firmware, the board is new and I can't find almost any content, I'm from Brazil and I would like this material to grow up in my country. Someone could send a firmware that works for SKR pro v1.2 + TMC 2209 UART with SENSORLESS + TFT24 (this I got a list of functional commands) and BLtouch. The parameters I can define. I am a beginner user and I have been working on this 3d printer for 3 months and have not been able to print anything so far

I'll leave my contacts below:
email: petinho.guto@gmail.com
WhatsApp: +55 (43) 999970444
facebook: https://www.facebook.com/peterson.carvalho.90/

myxxx323, 2020-12-26:
Для того что бы скомпилировалась прошивка, нужно изменить - TMCStepper@<1.0.0 на - TMCStepper@>=0.7.1