bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #98: Can't use TMC2208 in UART mode - can missing UART resistors be the cause?

TheDark, 2020-02-02:
As posted by @AnHardt in https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/issues/70#issuecomment-551315083

> The places for the resistors R61, R65, R67, R69, R70, R71 seem to be not populated on all boards.

I have such an unpopulated board. Can the missing resistors be the cause for the TMC2208 drivers not to work, causing "Error: ALL LOW"? Can I just add the resistors to make it work?

The 2208s work perfectly well on a SKR 1.3. Using latest Marlin bugfix 2.0.

[image]

[image]

[image]

blanosko, 2020-02-03:
You can take a look on schematic and figure out if there should be a resistor:

You can see that there should be 1K resistors. R70 is for X UART and R71 is for Y:
[image]

Here are for Z, E0, E1, E2:
[image]

GadgetAngel, 2020-02-04:
@TheDark and @blanosko Here is a picture of my newly bought SKR PRO V1.1 board, which DOES NOT have the R70, R71, R61, R65, R67, or R69 solder on the board. So without those resistors in place how can UART mode be working?  I connected up my TMC2209 V1.2 to the SKR PRO board and I can turn my stepper motors with TMC2209 in UART mode. How can this work if they are truly NOT connected?? I am missing something here. The TX line have to be connected

blanosko, 2020-02-04:
@GadgetAngel Your motors are working because UART have nothing to do with that. Drivers are controlled through EN, STEP/DIR pins.

UART on these drivers is used to **configure** them (How much current going to motors, Stealthchop treshold, monitoring temperatures. etc).

So we can elaborate:
I have MKS Gen 1.4 board. If I want to use TMC2208 drivers (blk.me branded) I have tu use Y splitter with 1K resistor. Basicaly thats what is in schematics of SKR Pro.
[image]

So if I disconnect TX line I could only read data from driver not write them. So Iam assuming that if there is no TX line resistors in place you can only read them.

theopensourcerer, 2020-02-04:
@blanosko 
> So if I disconnect TX line I could only read data from driver not write them. So Iam assuming that if there is no TX line resistors in place you can only read them.

Not sure if this is relevant or not, but when using Klipper firmware on my MKS Gen 1.4 you only need a single wire and no resistor to both read from and write to the 2208.

blanosko, 2020-02-04:
@theopensourcerer 
I was searching for some info about this and aparently it is possible with one wire without resistors. But Iam not sure when it can work and when not. So it would be great if someone with greater knowledge can sort this for once and for all.

theopensourcerer, 2020-02-04:
@blanosko I know it works - I've been using it for nearly a year with no issues. 

I recently bought this SKR Pro V1.1 board and am going to use some 2130s I have in SPI. I've been testing it on my bench and I can talk to them OK. Again I am using Klipper f/w.

theopensourcerer, 2020-02-04:
The point is I don't believe it's anything to do with hardware. If Klipper can do it on any random MCU board then so should Marlin presumably.

GadgetAngel, 2020-02-04:
> @GadgetAngel Your motors are working because UART have nothing to do with that. Drivers are controlled through EN, STEP/DIR pins. And they are connected in UART mode, but RX pin only.
> 
> UART on these drivers is used to **configure** them (How much current going to motors, Stealthchop treshold, monitoring temperatures. etc).
> 
> So we can elaborate:
> I have MKS Gen 1.4 board. If I want to use TMC2208 drivers (blk.me branded) I have tu use Y splitter with 1K resistor. Basicaly thats what is in schematics of SKR Pro.
> [image]
> 
> So if I disconnect TX line I could only read data from driver not write them. So Iam assuming that if there is no TX line resistors in place you can only read them.
> 
> EDIT: OR, if it works (reading and writing), then maybe they are using some integrated resistor in microcontroller or schematics are wrong (outdated) and dont correspond to the real board.

Ok, so if I understand this correctly. I can not configure my TMC2209 to run at different Microsteps or switch between Stealthchop and Spreadcycle on the file because the TX pins are not connected?  Has anyone tested this out?  What mode are my Drivers running in? When I did a M122 command during my test, all reagister coming back from the TMC2209 looked OK, and my setup for the driver was what I set it to be. But I could have gotten lucky because I only set it for 1/16th microstepping. I need to go back and see if I can set the parameters differently. I will do that tomorrow and report back what I found. So if I find I can not adjust parameters then SPI mode is the only way one can use the SKR PRO?

blanosko, 2020-02-04:
If you can see output of M122 then connection is OK. You can test if you can change say motor current, save it, reboot board and then try M122 and check if the settings are the new one.

Refer to what @theopensourcerer said. (One wire UART)

theopensourcerer, 2020-02-04:
@GadgetAngel The 2208/9 are supported by Klipper firmware on the SKR Pro. They only need one pin defined for the UART:

https://github.com/KevinOConnor/klipper/blob/master/config/generic-bigtreetech-skr-pro.cfg#L106

GadgetAngel, 2020-02-05:
> @GadgetAngel The 2208/9 are supported by Klipper firmware on the SKR Pro. They only need one pin defined for the UART:
> 
> https://github.com/KevinOConnor/klipper/blob/master/config/generic-bigtreetech-skr-pro.cfg#L106

I found it in the Datasheets for the TMC2208/9: 
"This mode (UART) allows replacing all control lines like ENN, DIAG, INDEX, MS1, MS2, and analog current setting VREF by a single interface line" https://www.trinamic.com/products/integrated-circuits/details/tmc2209-la/:
[image]

spattinson, 2020-02-07:
The pins file for the SKR Pro defines the following for UART:
```
#define X_SERIAL_TX_PIN  PC13
  #define X_SERIAL_RX_PIN  PC13

  #define Y_SERIAL_TX_PIN  PE3
  #define Y_SERIAL_RX_PIN  PE3

  #define Z_SERIAL_TX_PIN  PE1
  #define Z_SERIAL_RX_PIN  PE1

  #define E0_SERIAL_TX_PIN PD4
  #define E0_SERIAL_RX_PIN PD4

  #define E1_SERIAL_TX_PIN PD1
  #define E1_SERIAL_RX_PIN PD1

  #define E2_SERIAL_TX_PIN PD6
  #define E2_SERIAL_RX_PIN PD6
```
Clearly the same pin is used for TX and RX so is this an issue or not?
If I "grep X_SERIAL_ *" in the pins/stm32 to see if other vendors have the same setup:
```
scott@n550jk:~/pio/Marlin-bugfix-2.0.x/Marlin/src/pins/stm32$ grep X_SERIAL_ * 
pins_BTT_BTT002_V1_0.h:  #define X_SERIAL_TX_PIN  PE2
pins_BTT_BTT002_V1_0.h:  #define X_SERIAL_RX_PIN  PE2
pins_BTT_GTR_V1_0.h:  #define X_SERIAL_TX_PIN  PC14
pins_BTT_GTR_V1_0.h:  #define X_SERIAL_RX_PIN  PC14
pins_BTT_SKR_E3_DIP.h:  #define X_SERIAL_TX_PIN  PC10
pins_BTT_SKR_E3_DIP.h:  #define X_SERIAL_RX_PIN  PC10
pins_BTT_SKR_MINI_E3_V1_2.h:  #define X_SERIAL_TX_PIN  PB15
pins_BTT_SKR_MINI_E3_V1_2.h:  #define X_SERIAL_RX_PIN  PB15
pins_BTT_SKR_PRO_V1_1.h:  #define X_SERIAL_TX_PIN  PC13
pins_BTT_SKR_PRO_V1_1.h:  #define X_SERIAL_RX_PIN  PC13
pins_FLYF407ZG.h:  #define X_SERIAL_TX_PIN  PG13
pins_FLYF407ZG.h:  #define X_SERIAL_RX_PIN  PG13
pins_FYSETC_CHEETAH_V12.h:  #define X_SERIAL_TX_PIN  PA11
pins_FYSETC_CHEETAH_V12.h:  #define X_SERIAL_RX_PIN  PA12
pins_FYSETC_S6.h:  #define X_SERIAL_TX_PIN  PE9
pins_FYSETC_S6.h:  #define X_SERIAL_RX_PIN  PE8
pins_MKS_ROBIN_PRO.h:  #define X_SERIAL_TX_PIN  PF7
pins_MKS_ROBIN_PRO.h:  #define X_SERIAL_RX_PIN  PF8
```
We can see it is only BTT boards that have this feature on the stm32 platform, other vendors use different pins for tx and rx.

GadgetAngel, 2020-02-08:
@TheDark Which TMC2208 driver board are you trying to get to work? Can you post a picture of the driver boards’ top and bottom view?  Is it from FYSECT or BIQU?

TheDark, 2020-02-08:
> @TheDark Which TMC2208 driver board are you trying to get to work? Can you post a picture of the driver boards’ top and bottom view? Is it from FYSECT or BIQU?

GadgetAngel, 2020-02-09:
@TheDark post a picture of BIQU TMC2130 from the bottom and top, please

GadgetAngel, 2020-02-09:
> > @TheDark Which TMC2208 driver board are you trying to get to work? Can you post a picture of the driver boards’ top and bottom view? Is it from FYSECT or BIQU?
> 
> I believe it's a FYSECT clone, as it looks like those but is unmarked; pics below.
> 
> Also, I was wrong: the boards don't actually work in UART mode with the SKR 1.3, they just defaulted to dir/step mode and fooled me. The thing is, in step/dir mode on the pro many steps are skipped. Meanwhile, I've gotten some BIQU 2130s and also can't connect to them by SPI. I'm not sure if anything works at this point 😝
> 
> [image]
> 
> [image]
> 
> [image]

I am not familiar with THIS Chip. But from the picture I DO NOT see the UART Pin on the driver board. Can you send me a link to this driver board schematic?

TheDark, 2020-02-11:
Here they go!

[image]

[image]

blanosko, 2020-02-12:
You dont have a solder bridge on SPI pad at the bottom of the driver. It cant work in SPI mode if dont have that two pads soldered together.

GadgetAngel, 2020-02-12:
@blanosko if you look very closely he does have the SPI bridged together. BTT uses a 0 Ohm resistor that has a black part in the middle. @TheDark could you take another picture of the BTT TNC2130 but title the TMC2130 a bit so we can see if the two SPI pads are truly bridged together? I also need a shot for a manual I am working on

@TheDark Your jumpers on the SKR Pro V1.1 board for TMC2130 should look like this:
[image]

blanosko, 2020-02-12:
Yeah, you are correct. There is a resistor. My bad.

GadgetAngel, 2020-02-12:
@TheDark to get the TMC2130 to work in SPI mode the SPI jumper MUST NOT BE IN PLACE. 

[image]
[image]

GadgetAngel, 2020-02-12:
@TheDark it looks like you bought the TMC2130 in standalone mode. Double check, is there a 0 Ohm resistor across the two SPI pads. If there is one there then the driver board is set for Stand alone mode:  on page 4 of https://github.com/bigtreetech/BIGTREETECH-TMC2130-V3.0/blob/master/TMC2130-V3.0RM.pdf

blanosko, 2020-02-12:
So BTT 2130 have this inverted. You need open bridge for SPI and closed for STANDALONE. Iam I right? I have waterott drivers and they need solder bridge between pads near CS pin to work in SPI mode.

GadgetAngel, 2020-02-12:
@blanosko Here is the watterott documentation: https://learn.watterott.com/silentstepstick/pinconfig/tmc2130/

It show that the BTT is **THE SAME WAY**, the SPI jumper "CLOSED" is STAND ALONE mode and the SPI jumper "OPEN" is SPI Mode. 
[image]

TheDark, 2020-02-14:
Sorry for the delay in my answers, and thanks all for yours.

There is a jumper in the SPI pads, photos follow. If I remove it can I use the boards as SPI, or is it permanently in standalone mode?

[image]
[image]
[image]
[image]

GadgetAngel, 2020-02-15:
@TheDark you need to unsolder the jumper and then solder in CFG4 and CFG5 as the manual from BIGTREETECH says to. 
[image]

TheDark, 2020-02-15:
Thanks, I'll try that when I feel confident enough.

I'll probably have to remove the header to be able to remove the jumper without ruining the plastic and the pins. And those RG* pads are so tiny... the chances of me ruining the board will be high.

Regarding the 2208s, apparently Geeetech released some defective batches and maybe mine came from one of those. No serials to check though. The following video shows what happens when I connect them:

https://youtu.be/qIWuZ2AIt0E

closed 2020-03-19 by TheDark