bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #103: Sensor-less homing/probing not working SKR_PRO_V1.1 with TMC2209

fiveofds, 2020-02-10:
Sensor-less homing/probing not working SKR_PRO_V1.1 with TMC2209 I am unable to get the sensorless homing or probing working. I have good movement on all steppers and heating. Please advise.

fiveofds, 2020-02-10:
I would post my Configuration.h and Configuration.adv.h but I don't see how to.

fiveofds, 2020-02-10:
> /**
 * Marlin 3D Printer Firmware
 * Copyright (c) 2019 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (c) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <http://www.gnu.org/licenses/>.
 *
 */
#pragma once

/**
 * Configuration.h
 *
 * Basic settings such as:
 *
 * - Type of electronics
 * - Type of temperature sensor
 * - Printer geometry
 * - Endstop configuration
 * - LCD controller
 * - Extra features
 *
 * Advanced settings can be found in Configuration_adv.h
 *
 */
#define CONFIGURATION_H_VERSION 020004

//===========================================================================
//============================= Getting Started =============================
//===========================================================================

/**
 * Here are some standard links for getting your machine calibrated:
 *
 * http://reprap.org/wiki/Calibration
 * http://youtu.be/wAL9d7FgInk
 * http://calculator.josefprusa.cz
 * http://reprap.org/wiki/Triffid_Hunter%27s_Calibration_Guide
 * http://www.thingiverse.com/thing:5573
 * https://sites.google.com/site/repraplogphase/calibration-of-your-reprap
 * http://www.thingiverse.com/thing:298812
 */

//===========================================================================
//============================= DELTA Printer ===============================
//===========================================================================
// For a Delta printer start with one of the configuration files in the
// config/examples/delta directory and customize for your machine.
//

//===========================================================================
//============================= SCARA Printer ===============================
//===========================================================================
// For a SCARA printer start with the configuration files in
// config/examples/SCARA and customize for your machine.
//

[... 2240 lines cut ...]

/**
 * R/C SERVO support
 * Sponsored by TrinityLabs, Reworked by codexmas
 */

/**
 * Number of servos
 *
 * For some servo-related options NUM_SERVOS will be set automatically.
 * Set this manually if there are extra servos needing manual control.
 * Leave undefined or set to 0 to entirely disable the servo subsystem.
 */
//#define NUM_SERVOS 3 // Servo index starts with 0 for M280 command

// (ms) Delay  before the next move will start, to give the servo time to reach its target angle.
// 300ms is a good value but you can try less delay.
// If the servo can't reach the requested position, increase it.
#define SERVO_DELAY { 300 }

// Only power servos during movement, otherwise leave off to prevent jitter
//#define DEACTIVATE_SERVOS_AFTER_MOVE

// Allow servo angle to be edited and saved to EEPROM
//#define EDITABLE_SERVO_ANGLES

fiveofds, 2020-02-10:
> /**
 * Marlin 3D Printer Firmware
 * Copyright (c) 2019 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (c) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <http://www.gnu.org/licenses/>.
 *
 */
#pragma once

/**
 * Configuration_adv.h
 *
 * Advanced settings.
 * Only change these if you know exactly what you're doing.
 * Some of these settings can damage your printer if improperly set!
 *
 * Basic settings can be found in Configuration.h
 *
 */
#define CONFIGURATION_ADV_H_VERSION 020004

// @section temperature

//===========================================================================
//=============================Thermal Settings  ============================
//===========================================================================

//
// Custom Thermistor 1000 parameters
//
#if TEMP_SENSOR_0 == 1000
  #define HOTEND0_PULLUP_RESISTOR_OHMS 4700    // Pullup resistor
  #define HOTEND0_RESISTANCE_25C_OHMS  100000  // Resistance at 25C
  #define HOTEND0_BETA                 3950    // Beta value
#endif

#if TEMP_SENSOR_1 == 1000
  #define HOTEND1_PULLUP_RESISTOR_OHMS 4700    // Pullup resistor
  #define HOTEND1_RESISTANCE_25C_OHMS  100000  // Resistance at 25C
  #define HOTEND1_BETA                 3950    // Beta value
#endif

#if TEMP_SENSOR_2 == 1000
  #define HOTEND2_PULLUP_RESISTOR_OHMS 4700    // Pullup resistor
  #define HOTEND2_RESISTANCE_25C_OHMS  100000  // Resistance at 25C
  #define HOTEND2_BETA                 3950    // Beta value
#endif

#if TEMP_SENSOR_3 == 1000
  #define HOTEND3_PULLUP_RESISTOR_OHMS 4700    // Pullup resistor
  #define HOTEND3_RESISTANCE_25C_OHMS  100000  // Resistance at 25C
  #define HOTEND3_BETA                 3950    // Beta value
#endif

#if TEMP_SENSOR_4 == 1000
  #define HOTEND4_PULLUP_RESISTOR_OHMS 4700    // Pullup resistor
[... 3166 lines cut ...]
#if ENABLED(PRINTCOUNTER)
  #define SERVICE_WARNING_BUZZES  3
  // Activate up to 3 service interval watchdogs
  //#define SERVICE_NAME_1      "Service S"
  //#define SERVICE_INTERVAL_1  100 // print hours
  //#define SERVICE_NAME_2      "Service L"
  //#define SERVICE_INTERVAL_2  200 // print hours
  //#define SERVICE_NAME_3      "Service 3"
  //#define SERVICE_INTERVAL_3    1 // print hours
#endif

// @section develop

//
// M100 Free Memory Watcher to debug memory usage
//
//#define M100_FREE_MEMORY_WATCHER

//
// M43 - display pin status, toggle pins, watch pins, watch endstops & toggle LED, test servo probe
//
//#define PINS_DEBUGGING

// Enable Marlin dev mode which adds some special commands
//#define MARLIN_DEV_MODE

oldman4U, 2020-02-11:
What happens exactly when you home the axis? "Not working" can mean many things.

GadgetAngel, 2020-02-11:
@fiveofds To post your files, just post a LINK to them on your free google drive. 

Easy way: go to your computer find the folder that you use for Marlin. Right-click on the folder and Zip the folder. This will place all files in the Marlin folder into one Zip file. Now go to your web browser search for “google drive” Select “Google Drive” link from your search results. That link should now take you to your free google drive. Open up a file explorer window (on Windows machines use Win-E keys). Drag your Zip file (of the Marlin folder) over to your google drive page that is opened in your browser. The zip file will now be uploaded to your google drive. Go back to your browser and find the zip file on the google drive, then Right click the zip-file’s filename. Select “Get shareable link “. Now, google drive will turn on a shareable link for your zip file and place that link onto your computer’s clip board. Open any editor and choose Paste and you will see the shareable link to your zip folder. Now, come back here and paste that link into a post.  That’s how you POST your files.

fiveofds, 2020-02-11:
>   not working SKR_PRO_V1.1 with TMC2209  (#103) What happens exactly when you home the axis? "Not working" can mean many things.
>
> [
> {
> "@type": "EmailMessage",
> "@type": "ViewAction",
> "target": "https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/issues/103?email_source=notifications\u0026email_token=AHSZKOYFZ73XIVC2HHCFYX3RCJK5TA5CNFSM4KSTNIGKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOELLP2VQ#issuecomment-584514902",
> "url": "https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/issues/103?email_source=notifications\u0026email_token=AHSZKOYFZ73XIVC2HHCFYX3RCJK5TA5CNFSM4KSTNIGKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOELLP2VQ#issuecomment-584514902",
> "name": "View Issue"
> },
> "description": "View this Issue on GitHub",
> "publisher": {
> "@type": "Organization",
> "name": "GitHub",
> "url": "https://github.com"
> }
> }
> ]

If I send a M119 all the endstops report as open. When the end stop switches are pressed they still report as open. No matter what I do it stays open.When I home my printer it rises like I would expect it to do. When it reaches the top the steppers keep turning and make a grinding sound.I am using the latest Marlin-Bugfix-2.0.x Thank you MikeSent from my T-Mobile 4G LTE Device

oldman4U, 2020-02-11:
@fiveofds What do you mean with "When the end stop switches are pressed"?

GadgetAngel, 2020-02-11:
@fiveofds The ENDSTOP will show OPEN since you are using Sensorless Homing. You need to tune the Sensorless homing so it works. Please watch this Video on YouTube done by Teaching Tech on TMC2209 https://youtu.be/dOJbSrWVu_Q

Here is his video section that address Sensorless homing on the TMC2209 https://youtu.be/dOJbSrWVu_Q

Even tho his video is about the SKR V1.3, all the setting for Sensorless homing are the same. You will see how he tests for the bump values and the sensitivity settings to get Sensorless homing working

fiveofds, 2020-02-11:
The sensors for the end stops are triggered Sent from my T-Mobile 4G LTE Device

GadgetAngel, 2020-02-11:
@fiveofds The endstops MEAN nothing when using Sensorless homing! M119 will not show you the state of the endstops because you are NOT using ENDSTOP. You are setup for Sensorless Homing which uses the stepper motor driver to detect when you hit the beginning and ending of your axis travel

fiveofds, 2020-02-11:
Ok I get that, I set up sensorless homing in Marlin-Bugfix 2.0.x and it still does the same thing. Sent from my T-Mobile 4G LTE Device

fiveofds, 2020-02-11:
Ok I get that the sensors mean nothing. So then the problem is sensorless homing is not working. I have followed the instructions on YouTube and the sensorless homing is still not working. With the MKS PRO v1.1 do I need a jumper other than the one for UART mode? Sent from my T-Mobile 4G LTE Device

GadgetAngel, 2020-02-11:
@fiveofds The way you describe the motor behavior is EXACTLY as it should be.  The motors will grind when they hit the end of travel for the axis. You have to set the adjustment sensitivities properly to get the motors NOT to grind.  So I do not understand the problem. You have setup sensorless homing in Marlin but the same values for each printer will be unique.  

I looked at your Configuration_adv.h as best I could. Not having the ability to download the file into my editor and search for key variables makes helping you VERY hard.  Why do you have "#define X_HYBRID_THRESHOLD 0 // [mm/s]" set to 0?  You have StealthChop on but with this setting it appears that Marlin will always be in SpreadCycle Mode.  You problably want to adjust that up to at least to 100. See what that does. I might not have the expertise to help in your situation. What printer are you trying to get this working on? An HE3D is a delta printer is it not? I have zero experience on how to setup delta printers.  

But from your description, the printer is doing as it's suppose to do.  Why do you want to use sensorless homing? Why not just keep your endstops? 

Do this with the TMC2209 and keep your endstops: 

[image]

Your jumpers for configuring the TMC2209  in UART mode for the SKR Pro V1.1 should look like this:

[image]

Try NOT using Sensorless homing, just to make sure that the TMC2209 are working with your SKR PRO V1.1 board.  If that setup works then we need to try and figure out your sensorless homing settings

fiveofds, 2020-02-11:
Thank you, my jumpers are correct, yes the HE3D is a Delta printer. I
understand that you are telling me that the sensorless settings are most
likely the problem. I just changed the HYBRID_THRESHOLD to 200 and am
compiling Marlin now.

Mike

oldman4U, 2020-02-11:
GadgetAngel and oldman4U are the new support dream team,-)

@fiveofds I have a Marlin v2.0 firmware with sensor-less homing working fine in case you can not get it to work.

fiveofds, 2020-02-11:
I can not get it to work so far,
Can I send you a link to my configuration files to see if you can help me?
Mike

GadgetAngel, 2020-02-11:
@fiveofds Yes, send the link, @oldman4U  will look at it. I will also

fiveofds, 2020-02-11:
https://drive.google.com/open?id=13CkPc5H97uPpQNddQdDU2EwGaLRzWRsu

fiveofds, 2020-02-11:
I am thinking that I must be missing a setting somewhere for the sensorless
homing and probing. I so getting very frustrated

GadgetAngel, 2020-02-11:
@fiveofds who makes that printer? Does this printer have a default configuration file under Marlin?

fiveofds, 2020-02-11:
No there is no Marlin configuration I am it comes with a
different controller board, I switched it to the SKR PRO V1.1 because I
plaid to add the ability to use a second color later on.

Mike

GadgetAngel, 2020-02-11:
@fiveofds When you flashed your Marlin firmware, did you go into the menu and have it restore failsafe. That printer my not allow the changes you made to take effect until you do a restore failsafe.

fiveofds, 2020-02-11:
Yes I didSent from my T-Mobile 4G LTE Device

GadgetAngel, 2020-02-11:
@fiveofds try this:

> #if EITHER(SENSORLESS_HOMING, SENSORLESS_PROBING)
    // TMC2209: 0...255. TMC2130: -64...63
    #define X_STALL_SENSITIVITY  50 // was set at 250 might be too high giving you a False positive
    #define X2_STALL_SENSITIVITY X_STALL_SENSITIVITY
    #define Y_STALL_SENSITIVITY  50  // was set at 250 might be too high giving you a False positive
    #define Z_STALL_SENSITIVITY  50  // was set at 250 might be too high giving you a False positive
    //#define SPI_ENDSTOPS              // TMC2130 only
    #define IMPROVE_HOMING_RELIABILITY
  #endif

fiveofds, 2020-02-11:
>   not working SKR_PRO_V1.1 with TMC2209  (#103) @fiveofds try this:

Do I need to run a wire from the Diag pin on the 2209 to the SKR Pro v1.1 sensor pin for the end stop? Sent from my T-Mobile 4G LTE Device

fiveofds, 2020-02-11:
That did not work

GadgetAngel, 2020-02-11:
@fiveofds Does you TMC2209 have a Diag pin going up or down on the driver board? if the Diag pin is going up, then yes you need to connect the Diag pin to the endstop. But if the Diag pin is going down then NO, the Diag pin is connected to the endstop for each Axis when the Diag pin is facing down. If you Diag pin is missing on the TMC2209 then you need to solder a pin in and place if facing down.

You may have to go to the HE3D facebook group to see if anyone else there has gotten this to work on your printer. I am running out of ideas except for one more:

Leave the other change in place for stall_sensitivity and change these:
[image]

GadgetAngel, 2020-02-11:
@oldman4U i am running out of ideas can you help?

oldman4U, 2020-02-11:
Hello fiveofds

@GadgetAngel Help is on the way I hope!

@fiveofds First and maybe most important. In case you use this printer to make money - do not play around with it. If it is a hobby, don't let it frustrate you. See my long ticket related to A4988 motor drivers.... Its a hobby!;-))))

Using the SKR Pro with TMC 2209 v1.3 stepper drivers and Sensorless homing does not require any cabling on the motherboard. Everything is done internally like shown in all the YouTube videos.

Please find attached my configuration files and please let me know in case you have problems compiling them. There are also some settings related to BLTouch, but sensorless homing should work anyway.

Hope this helps
[SKR v1.1 Pro Marlin v2 Firmware.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/files/4187615/SKR.v1.1.Pro.Marlin.v2.Firmware.zip)

oldman4U, 2020-02-11:
Sorry I forgot to explain. Grindelwald is the code name of the printer I build based on the SKR Pro. Just in case you find the name in the firmware in some areas.

Everything is black with a few copper parts and some white elements on top.

fiveofds, 2020-02-11:
Thank you, I will give it a try now

fiveofds, 2020-02-11:
Question Grizwald is not a Delta printer is it? But I think sensorless
homing is working I believe the problem is in the latest Marlin-BugFix
2.0.x

oldman4U, 2020-02-11:
No, Grindelwald is not a Delta but a Cartesian.

This is great news, thank you!

I would check now line by line what the difference is and to compile it every few lines you changed. Doing so you should be able to find the problem(s). If you want, I can compile the same setup without BLTouch but with the official v2.0.3 of Marlin. Please let me know if you would like to give it a try.

fiveofds, 2020-02-11:
>   not working SKR_PRO_V1.1 with TMC2209  (#103) No, Grindelwald is not a Delta but a Cartesian.

Sensor less homing is definitely working for me. I have adjusted it to my printer. I am using the most current version of Marlin not the bugfix version. There seems to be a problem with the bugfix version that was the issue. You can try to get the bugfix working if you want. Thank youMikeSent from my T-Mobile 4G LTE Device

GadgetAngel, 2020-02-11:
@fiveofds if you problem is solved you might want to close the ticket. I did not know bugfix had bugs, I think I will switch over to version 2.0.3 also. Thanks @fiveofds for that information and thank you @oldman4U for helping out

fiveofds, 2020-02-12:
I found that when  X_HOME_DIR, Y_HOME_DIR, or  Z_HOME_DIR   is changed from -1 to 1 sensorless homing stops working. For a delta printer, it has to be set to 1.
 
// Direction of endstops when homing; 1=MAX, -1=MIN
// :[-1,1]
#define X_HOME_DIR  -1  //MPD
#define Y_HOME_DIR  -1  //MPD
#define Z_HOME_DIR  -1  //MPD

So my problem is still not solved.

oldman4U, 2020-02-12:
Well, i am not a Delta expert, nor a Marlin scientist. But i would try to change a few homing related settings like the MIN_PLUG to MAX_PLUG to see if this solves the problem.

Also this setting could make a difference i guess:
// Manually set the home position. Leave these undefined for automatic settings.
// For DELTA this is the top-center of the Cartesian print volume.
//#define MANUAL_X_HOME_POS 0
//#define MANUAL_Y_HOME_POS 0
//#define MANUAL_Z_HOME_POS 0

Please let us know and don't forget, it is a hobby;-)

fiveofds, 2020-02-13:
I have been trying different settings, what I do know is that if I have Marlin configured as a non-delta printer sensorless homing works. But when I configure it at a delta printer sensorless homing does not work. I think this is an issue with Marlin, not the SKR_PRO V1.1 or the TMC2209. Maybe this issue should be moved to the Marlin track instead of the Bigtreetech forum. What do you think?

oldman4U, 2020-02-13:
Hi fiveofds.

Hmmm, it almost sounds like a Marlin issue, so i an afraid at least i can not help further, sorry. Maybe there is someone else who has a Delta and can test this.

GadgetAngel, 2020-02-13:
@fiveofds From the above you have the following settings: 
`#define X_MAX_ENDSTOP_INVERTING false

#define Y_MAX_ENDSTOP_INVERTING false

#define Z_MAX_ENDSTOP_INVERTING false `

For Sensorless homing to work, it’s my understanding that these signals need to be inverted, please try this:

#define X_MAX_ENDSTOP_INVERTING true

#define Y_MAX_ENDSTOP_INVERTING true

#define Z_MAX_ENDSTOP_INVERTING true

Please let me know if that worked.

fiveofds, 2020-02-13:
>   not working SKR_PRO_V1.1 with TMC2209  (#103) @fiveofds From the above you have the following settings:

When I try to compile it I am told that SENORLESS HOMEING requires X_MAX_ENDSTOP_INVERTING = false when homing TMC2209 to X_MAXMikeSent from my T-Mobile 4G LTE Device

fiveofds, 2020-02-13:
I can get sensorless homing working IF it is configured as a cartesian printer but then configuring for my delta printer:

// Direction of endstops when homing; 1=MAX, -1=MIN
// :[-1,1]
#define X_HOME_DIR 1 //MPD
#define Y_HOME_DIR 1 //MPD
#define Z_HOME_DIR 1 //MPD

When this is switched from -1 to 1 sensorless homing stops working
Any advice?

oldman4U, 2020-03-10:
Hello Mike.

Could you please confirm that "stop working" means that the motor spins for a second or so, but stops after that and never homes correctly? Seems i can reproduce the issue after changing the way the motors are connected to my printer.

fiveofds, 2020-03-10:
Yes, it isMikeSent from my T-Mobile 4G LTE Device

  not working SKR_PRO_V1.1 with TMC2209  (#103) Hello Mike.

oldman4U, 2020-03-10:
OK, so i have to do more tests, but it seems that i can reproduce the issue on my machine. I will do some more tests and let you know.

Would you be so kind and reopen this ticket. Maybe someone else can help or this issue has to be escalated to the developers.

fiveofds, 2020-03-10:
Ok, I will try to figure out how to do that.Sent from my T-Mobile 4G LTE Device

fiveofds, 2020-03-10:
Issue still exists

oldman4U, 2020-03-11:
Hi.

I can confirm that i can see the issue you described on my machine on all axis as soon as i change the homing direction from -1 to 1. Using the latest Marlin version 2.0.4-4 with my SKR Pro v1.1 with BTT 2209 v1.3 and a TFT28 display if this is important. Fortunately i have a working solution because my system is using -1 homing direction, but for me this looks like something which could probably be fixed in Marlin. I do not know how to escalate this to the developers, but please let me know in case i can help somehow.

guruathwal, 2020-03-14:
> I can get sensorless homing working IF it is configured as a cartesian printer but then configuring for my delta printer:
>
> // Direction of endstops when homing; 1=MAX, -1=MIN
> // :[-1,1]
> #define [...]

@fiveofds To Summarize your issues and findings:
You are trying to enable sensorless homing on a Delta printer
The sensorless homing is working perfectly if it is configured as a cartesian printer.

With this information, I can only think of one possible reason :

the Diag PINs of the TMC2209 drivers are connected to the `MIN Endstops` connectors on the board, which works with the printers with homing direction towards `-1(MIN)`. But for the delta-printer, the homing direction is `1 (MAX)`. The Delta settings might be forcing the firmware to use the `MAX Endstops` rather than the `MIN Endstops`. So you need to check the PIN configuration for the Delta-printer settings for your board to check if this is the case.

If there is any other information/point I missed. please let me know because this is a very long page to scroll and had to read it backward from bottom to top.

fiveofds, 2020-03-14:
With the SKR Pro v1.1 the diag pins are connected internally on the board. I don't know about the pins file for the SKR Pro v1.1. Sorry I am not a Marlin high speed guy. I am just trying to get my printer to work. But I believe that you have the right idea about what is happening Sent from my T-Mobile 4G LTE Device

https://drive.google.com/file/d/1LUrxH_01t1XlZNFe2P4wqudVntanQp1f/view?usp=sharing

guruathwal, 2020-03-14:
**This information form the Marlin documentation page confirms my theory:**

[image]

https://marlinfw.org/docs/configuration/configuration.html#endstop-plugs

guruathwal, 2020-03-14:
**@fiveofds  according to SKR_PRO_V1.1 Pinout, the StallGuard pins are only connected to the `PB10`, `PE12` & `PG8` PINS**

[image]

[image]

**you have also used the Max End stops**

*You need try swapping the MIN & MAX pin in your board-specific file*

`Marlin\src\pins\stm32\pins_BTT_SKR_PRO_V1_1.h` (Line No. 44 to 49) 

**From this:**
```c
#define X_MIN_PIN          PE10
#define X_MAX_PIN          PB15
#define Y_MIN_PIN          PE12
#define Y_MAX_PIN          PE10
#define Z_MIN_PIN          PG8
#define Z_MAX_PIN          PG5
```
**To this:**

```c
#define X_MIN_PIN          PE15
#define X_MAX_PIN          PB10
#define Y_MIN_PIN          PE10
#define Y_MAX_PIN          PE12
#define Z_MIN_PIN          PG5
#define Z_MAX_PIN          PG8
```

guruathwal, 2020-03-15:
@fiveofds.  Can you please edit or remove your second and third comment? because Its very frustrating to scroll on phone.

oldman4U, 2020-03-15:
My Guru, my hero.

What you recommended works for me;-) THANK YOU

Hope it also works for fiveofds.

For me this looks more like a workaround than a solution. Is this right?

guruathwal, 2020-03-15:
@oldman4U I am Glad to know that it worked :)
This is the only possible way to get it working.

oldman4U, 2020-03-16:
@guruathwal Maybe it would be a good idea to have this saved somewhere as a kind of "How to" article. There are users like you, familiar to read the documents and getting the things done right. But for someone new to this it is hard to understand. I am sure there are more users out there having a similar problem and the solution is easy, once you know it - like always;-)

Thank you again, i really appreciate your help

@fiveofds Hi Mike. Have you changed the pins file and does your printer work now?

Please let us know

fiveofds, 2020-03-16:
Well now I am having another issue but I'll get it working soon. Thanks Sent from my T-Mobile 4G LTE Device

oldman4U, 2020-03-16:
I know this problem - and she is standing between you and your printer 🤪

Georg

guruathwal, 2020-03-18:
@oldman4U Another way to get this working is by cutting/removing the Stallguard pin on the TMC board, as shown in the below image. then it can be connected to the Max Endstop pins with a jumper wire. this does not require any additional firmware change.

[image]

oldman4U, 2020-03-18:
Hi.

Yes, but then i can not use senseless homing which works fine.

[6 more replies]