bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #104: Did I just brick the mcu?

Nyxtegida, 2020-02-11:
I recently received my skr-pro-1.1 and from the start it wouldn't respond to pronterface commands.
The board was and still is empty, not even drivers attached.
I tried compiling a new firmware, write it on the sd card, and still nothing.
It was detected as a usb-serial device, connected at 250000 baud rate in pronterface, but failed to respond to any command.
After a few tries with different firmware, it wouldn't even show up when connected to the pc.
Then I tried st-link.
All firmware I used with the st-link were from here: https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0/issues/20
I was reading something from the mcu, so I tried to write the firmware directly with the st-link.
I got a download complete, but no confirmation from stmcube, and when I tried to read again everything was FFFF.
I also tried to re-write the bootloader in case I deleted it somehow, and I got the same result.
There.

GadgetAngel, 2020-02-11:
@Nyxtegida see closed issue https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/issues/72

There is a firmware file to use with ST-Link that might get you back up and running. You will see a link to download the firmware file from one of the conversions. So please read all the post comments to find it

Nyxtegida, 2020-02-11:
I already tried all the firmware, but the problem is that after flashing I get an error that it couldn't verify. Every firmware I tried, the memory banks red FFFFFFFF. As if it didn't write. Otherwise the chip was detected fine.

GadgetAngel, 2020-02-12:
@Nyxtegida i would send back the board @bigtreetech @yangwenxiong

Nyxtegida, 2020-02-22:
Well. It eventually died. I think now its on a permanent state of reset, or that what the st-link says. I got a refund for it so I'm getting another.

closed 2020-02-22 by Nyxtegida