bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #106: flashing failures & bootloader support for long file names

spattinson, 2020-02-14:
I have been reflashing marlin quite a lot recently and have run into issue where I have flashed firmware and changes I have made to marlin have made no effect to printer. The printer looks like it has flashed, the led blinks.
What I believe is happening:
Access the sdcard on your computer, copy firmware.bin to it - your computer creates a FAT long filename for the file despite the fact that the filename is exactly 8.3 format.
Insert card in 3d printer and reset, flashing occurs and firmware.bin is renamed FIRMWARE.CUR by bootloader. The long filename entry is left alone because the bootloader does not support LFN so it becomes orphaned.
Repeat the above a few times and flashing no longer works properly, the filesystem entry for firmware.bin is pointing to the wrong file, probably FIRMWARE.CUR so you end up flashing the same firmware you already had.
Remove card and check file system:
```
scott@n550jk:~/pio/Marlin/.pio/build/BIGTREE_SKR_PRO$ sudo fsck /dev/mmcblk0p1 
fsck from util-linux 2.31.1
fsck.fat 4.1 (2017-01-24)
Orphaned long file name part "firmware.bin"
1: Delete.
2: Leave it.
? 1
Orphaned long file name part "firmware.bin"
1: Delete.
2: Leave it.
? 1
Perform changes ? (y/n) y
/dev/mmcblk0p1: 4 files, 101/974732 clusters
```
Recent versions of marlin default to using  the external SD card if you have one so your gcode and eeprom.dat will be on there. the microsd on the board is only used for flashing.
Avoid this problem in a number of ways:
On linux you can ensure the sdcard is mounted "msdos" type, this forces 8.3 format only without long filenames.
regularly reformat the sdcard to get rid of errors.
run check filesystem for errors, or fsck on linux.
update "#define STRING_CONFIG_H_AUTHOR " in configuration.h each time you make changes and check with M115 after flashing - I have not verified this yet, I'd like to see a date/time stamp maybe easier.

Further background info: Long filenames on FAT filesystems are supported by a look up table, for compatibility the 8.3 filename still exists so apps that dont support LFN can still use the sdcard. The look up table maintains cross reference between 8.3 and long filenames.

vover000, 2020-04-01:
Помогите решить проблему.
Дело в том, что при сборке 3Dпринтера, случайно закоротил  проводом нижнюю часть платы SKR PRO v1.1, в результате чего сгорел Микроконтролер  Stm32f407zgt6 который установлен на ней, я нашёл и купил новый Stm32f407zgt6, заменил его.
Теперь проблема прошить правильный файл bootloader.bin.
Проще говоря нужен правильный файл bootloader.bin для этого Микроконтролера.
Материнская плата skr pro сама не пострадала от короткого замыкания, но прошивка Marlin-2.0
прошитая на прямую через uart не работает без этого файла bootloader.bin.

vover000, 2020-04-01:
Help me solve the problem.
The fact is that when assembling a 3dprinter, I accidentally shorted the lower part of the SKR PRO V1.1 Board with a wire, resulting in the burning of the stm32f407zgt6 Microcontroller that is installed on it, I found and bought a new Stm32f407zgt6, replaced it.
Now the problem is to flash the correct bootloader file.bin.
Simply put, you need the correct bootloader file.bin for this Microcontroller.
The skr pro motherboard itself was not affected by a short circuit, but the Marlin-2.0 firmware
stitched directly via uart does not work without this bootloader file.bin

AnHardt, 2020-04-01:
https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/issues/25#issuecomment-607009288