bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #109: Can't find PSU_Control Pin

derjg, 2020-02-21:
I want to use the auto power down funktion, but can't find the pin and funktion.
Couldyou please help me?

rromanii, 2020-02-27:
It took me a while to find out but it looks like the official relay uses PF15. 

https://github.com/bigtreetech/BIGTREETECH-Relay-V1.2/blob/master/BIGTREETECH%20Relay%20V1.2/BIGTREETECH%20Relay%20V1.2/Relay%20V1.2-Operating%20Instruction.pdf

That means that it takes the pin for the ESP01 wifi module GPIO 2. 
[image]

The relay manual states that you should be able to uncomment a line to get it working but I don't see in the pins file.

The good news is you should be able to add a define PS_ON_PIN to the pins_BTT_SKR_V1.1.h file and assign it to any pin you want.  You just have to figure out a pin that isn't being used. Just remember to uncomment define PSU_CONTROL in the configuration.h file as well.

derjg, 2020-03-03:
Thanks a lot, it works fine now :-)

Great Service 

 
 

Gesendet: Donnerstag, 27. Februar 2020 um 23:50 Uhr

https://github.com/bigtreetech/BIGTREETECH-Relay-V1.2/blob/master/BIGTREETECH%20Relay%20V1.2/BIGTREETECH%20Relay%20V1.2/Relay%20V1.2-Operating%20Instruction.pdf

bobgreen64, 2021-12-28:
This does not make sense.  By using the PF15 then you cannot have WiFi module.  The Marlin default is PF13 but that is not shown in any of the pintouts.  

I manage to make the relay work on SKR Turbo and selected the pin I wanted to use (P1_00) .  I am unable to make it work on my BTT SKR Pro.

I should be able to assig a specific pin for that function

rromanii, 2021-12-29:
On marlin you should be able to define any pin you want to the PSU_PIN regardless of the board you are using. 

If redefining PSU_PIN doesn't work then you might want to check the board pins file to see if the pin it isn't being used for something else or PSU_PIN isn't being redefined to a different pin.