bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #110: SKR Pro v1.1 + TMC5160 + Double X + SensorLessHoming + Motor 24V 1.8deg 2.0A ?...

Uri-34, 2020-02-25:
Good day ! 
I ask for help in solving the problem !!! 
My kit : SCR Pro v1. 1 & TMC5160 x 6 & Motor 24V 1.8 deg 2.0 A x 5 & Motor 24V 1.8 deg 1.5 A & Motor 24V 1.8 deg 3.0 A & Marlin 2.0 ! 
My task: control 4 motors : X 2Y Z, and two more E1 E2 for my purposes ! I will be very grateful for any help in solving the problem !

Uri-34, 2020-02-25:
Here is a link to the configuration files : https://yadi.sk/d/gGtXFX3MJqUf6Q
I will be grateful for pointing out the wrong points !

tphan26363, 2020-02-25:
I can't point out what is wrong  since I don't have time.  But here is a configuration with mixed TMC2208 and TMC5160 with Dual Z.  You can use it as an example. it should not be too hard.  Just to make sure the jumper on the board set accordingly for your TMC( serial mode or SPI mode)

[Configuration.h.txt](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/files/4252013/Configuration.h.txt)
[Configuration_adv.h.txt](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/files/4252014/Configuration_adv.h.txt)
[pins_BIGTREE_SKR_PRO_V1.1.h.txt](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/files/4252015/pins_BIGTREE_SKR_PRO_V1.1.h.txt)

Uri-34, 2020-02-27:
Good day ! Unfortunately, the firmware with your configuration files is not compiled ! Maybe you didn't send all the changed files ?

Uri-34, 2020-02-27:
I would like to understand the nuances of configuring TMC drivers... ! :)

tphan26363, 2020-03-19:
it just an example .  it can not apply to your case.  But  I mark all the
places that need to change.  you can follow that and do your own.  The
hardware detail is not the same

Uri-34, 2020-03-19:
Good day ! 

I've dealt with everything except sensorless ! 
Who can tell if you need to solder DIAG1 on TMC5160 ? 
As far as I understand this is a necessity !?

Best regards.

Uri-34, 2020-03-20:
Good afternoon! All the existing problems have been solved for this day ! :) Thank You ...

Uri-34, 2020-03-20:
Go on ...

oldman4U, 2020-06-08:
Hello Uri-34.

Could you please close this ticket.

Thank you