bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #111: SKR PRO 1.1 + TCM5160 with debug pin added "Bad Response"

yakumoklesk, 2020-03-04:
Hi. I have downloaded the bigtreetech branch for SKR-PRO-V1.1. I've soldered a pin on my TCM5160 diag hole. I am using TCM5160 driver mode, not standalone, and configured the pins. Also had set the TCM library dependency for the repository, although for some reason keeps downloading another that is version 0.6.2. In case I "force" to use the 0.4.5 from the repo, the compilation fails instantiation TCM5160Stepper constructor.

I really do not know what to do. I at least would like to know if in order to get a correct response to the M122 comand in the pronterface, I need to have the motors connected to the board, or is only enought to have the drivers and the 24 volts connected into Power and Motor power.

Thanks in advance.

[PlatformAndConfig.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/files/4287757/PlatformAndConfig.zip)

oldman4U, 2020-06-08:
Hello yakumoklesk.

Have you been able to solve the reported issue in the meantime?

Please let us know, thank you

oldman4U, 2020-06-15:
@bigtreetech Could you please close this unneeded ticket. Thank you

evandene, 2020-11-25:
Dear yakumoklesk,
where you able to solve the TMC5160/SKR Pro problem?
If so, can you tell me how?