bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #112: SKR PRO V1.1 + TMC5160. Random freeze while printing from USB/TFT

jukillo, 2020-03-08:
Hello mates,

I have been looking for a solution to this problem for two months.

When I try to print from USB or from BTT TFT24, the printer stops randomly. Motors on, heaters too: bad situation. Printing from S3D i receive Firmware unresponsive error when this happens. If i click on force next, the printer continues.

The only way to finish a job is printing from LCD 12864 SD.
I have tested several buffer configurations, thinking that this is the problem, but nothing changes. Several baudrates too.
Same problem with different versions of marlin 2. (now updated to 2.0.4.4)

Any idea?

Thanks.
Regards.

LegendAlexander, 2020-04-09:
@jukillo 
what do you mean "print from usb" I thought the usb drive didn't work. How did you get it working?

davidmarli, 2020-04-22:
same problem here !! 
Utmost printing file stop at a moment.
Where do you find the "force next" option ?

jukillo, 2020-04-22:
> @jukillo
> what do you mean "print from usb" I thought the usb drive didn't work. How did you get it working?

Printing from USB = Printing from computer.

@davidmarli , force next option is showed in S3D when you print via USB (from computer).

tphan26363, 2020-04-25:
I having the same problem.  Print Ok using SD card on TFT-35. When print over USB port it's randomly stop and Octorprint report no communication.  How ever I can see that the temperature still report correctly.  Use fake acknowledgement, I some time can get it moving again .  How ever it always report:
Error:Line Number is not Last Line Number+1, Last Line: 132
This is look like  the communication over USB port was having a problem and can't  get correct information from the USB port.  Also I can some time see that the printer suddenly trying
to move to a position that way out side the print bed...  This could be a result from noisy USB combine with bad error checking or timing problem in the planner

jukillo, 2020-04-25:
At least you can print in standalone using TFT-35.. Not I..

daviddominguez, 2020-05-09:
I have the same issue. Have you found the problem?

jukillo, 2020-05-10:
Nope.
I have given up.. and i know that more people have this issue, but nobody have solve it.

tphan26363, 2020-05-11:
I use a very bad and long usb cable.. and now it is not stoping any more. Good and high quality cable don’t work. Short cable don’t work. Now it’s time to look for cheap low quality cable...

jukillo, 2020-05-11:
I know that my problem is BTT touch communication. This is freezing uart communications randomly.

Printing from 12864 lcd, zero problems.

daviddominguez, 2020-05-11:
Im not using any TFT touch screen plugged on SKR 1.1, just a basic reprap discount screen, but even with this setup, USB connection dies (always) when I start printing through USB from simplify3D or randomly when connected. At least I can use SD card.

jukillo, 2020-05-12:
> Im not using any TFT touch screen plugged on SKR 1.1, just a basic reprap discount screen, but even with this setup, USB connection dies (always) when I start printing through USB from simplify3D or randomly when connected. At least I can use SD card.

SKR pro? what marlin version? What buffer settings you have?

daviddominguez, 2020-05-12:
> SKR pro? what marlin version? What buffer settings you have?

Yes, SKR 1.1 Pro with default buffer setting on bugfix-2.0.x branch

phlebaspho, 2020-05-26:
In my case, after much A/B testing I realised it was not the TFT35 or pronterface that was losing connection - it was "device connected to Marlin serial port 2" that experienced the the comms loss. Moving the host device (in my case TFT35-E3) to the first serial port in Marlin config.h fixed it.

jukillo, 2020-05-26:
> In my case, after much A/B testing I realised it was not the TFT35 or pronterface that was losing connection - it was "device connected to Marlin serial port 2" that experienced the the comms loss. Moving the host device (in my case TFT35-E3) to the first serial port in Marlin config.h fixed it.

Hum... interesting this.
How did you test it? I will check it this days..

Thanks for info!

jukillo, 2020-05-28:
Seems to work..

#define SERIAL_PORT 1
#define SERIAL_PORT_2 -1

I only tried prints of 1,5h about, but all them finished well.

phlebaspho, 2020-05-29:
I lucked into finding a particularly pathological piece of gcode that would trigger the hang up after about 7 minutes.

Then it was a case of change everything relevant trying to find anything that made it better or worse.

I'm now at 100+ hours of printing without that issue recurring (plenty of other issues with being a bit too aggressive on movement speeds lately).

jukillo, 2020-05-29:
Finally seems that the problem is solved. Im still printing batch series of 1.5h - 5h but all jobs has finished well

Thanks phlebaspho!

daviddominguez, 2020-05-30:
> Finally seems that the problem is solved. Im still printing batch series of 1.5h - 5h but all jobs has finished well
> 
> Thanks phlebaspho!

Printing from USB or TFT?

jukillo, 2020-05-31:
> > Finally seems that the problem is solved. Im still printing batch series of 1.5h - 5h but all jobs has finished well
> > Thanks phlebaspho!
> 
> Printing from USB or TFT?

From TFT. 
I havent tried from USB yet.

daviddominguez, 2020-05-31:
> > > Finally seems that the problem is solved. Im still printing batch series of 1.5h - 5h but all jobs has finished well
> > > Thanks phlebaspho!
> > 
> > 
> > Printing from USB or TFT?
> 
> From TFT.
> I havent tried from USB yet.

Its good to know that at least you are able to print for TFT, let me know if you achieve to print from USB. I'm still stuck in this part

phlebaspho, 2020-05-31:
My experience is that I need a firmware  re-compile for that. I keep the TFT on port 1 by default - if I print over USB from pronterface/slic3r that device has the hang ups.

#define SERIAL_PORT -1
#define SERIAL_PORT_2 1

gives me print all day from USB.

#define SERIAL_PORT 1
#define SERIAL_PORT_2 -1

give me print all day from TFT.

Don't have a single build that swings both ways though.

daviddominguez, 2020-06-17:
> My experience is that I need a firmware re-compile for that. I keep the TFT on port 1 by default - if I print over USB from pronterface/slic3r that device has the hang ups.
>
> #define SERIAL_PORT -1 [...]

Thanks for your help, but I'm still stuck trying to print from USB, your config doesn't work for me. I'll keep trying combinations.

jukillo, 2020-06-17:
David, do you tried to print from usb disconnecting the tft screen?

daviddominguez, 2020-06-17:
> David, do you tried to print from usb disconnecting the tft screen?

No, should I disconnect it from cables and use the following configuration?:

#define SERIAL_PORT -1
#define SERIAL_PORT_2 1

Thanks in advance

jukillo, 2020-06-17:
If the problems come from the com ports, with this we are sure that only exists one open communication.

Try it. Lost to the river...

daviddominguez, 2020-06-17:
> If the problems come from the com ports, with this we are sure that only exists one open communication.
> 
> Try it. Lost to the river...

Same result :( this is quite frustrating.