bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #113: compiling problem

Tinchus2009, 2020-03-10:
Sorry to report this, not really used to handle comipling in VS code. I tried to compile the formware for the bigtreetech skr pro v1.1
I followed all the instructions I found here very carefully.
Compilation seems to finish ok (only yellow messages), but the final linking fails with this message:
c:/users/xxxx/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld.exe: cannot open linker script file stm32f407xg.ld: No such file or directory

Can you help me please?

janoc, 2020-03-10:
See #88, mainly the reply: https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/issues/88#issuecomment-570828632

Tinchus2009, 2020-03-11:
Thanks for the quick answer. It didnt worked. Is this board "stable". Something is not really good here: Doing that edit did not ork, but then I rolled back the changes, tried to compile again, and Im not being able to get to the point I got before. Now I receive the error almost inmediatly and it s not the same error as before.
Just to be sure I didnt touch anything else, I erased the folder, unziped the firmware again, I tried to compile again and now I received once more different errors, this time without even editing anything:

         ~^~~~~~~
*** [.pio\build\BIGTREE_SKR_PRO\src\src\HAL\HAL_STM32\fastio.cpp.o] Error 1
*** [.pio\build\BIGTREE_SKR_PRO\src\src\HAL\HAL_STM32\persistent_store_impl.cpp.o] Error 1
*** [.pio\build\BIGTREE_SKR_PRO\src\src\HAL\HAL_STM32\persistent_store_flash.cpp.o] Error 1
*** [.pio\build\BIGTREE_SKR_PRO\src\src\HAL\HAL_STM32\persistent_store_sdcard.cpp.o] Error 1

closed 2020-03-11 by Tinchus2009