bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #114: SKR PRO V1.1 +TFT35 V2.0 + 6 x TMC2130 SPI + Dual Z + Sensorless Homing

BrianS2K, 2020-03-10:
Hello,
I have 2 issues : 
1) I have 2 Z motors, I test them individualy (with the 2 jumpers) they works, but when I connect the 2 motors they make an horrible noise and don't move, just vibrate....

2) I enable Sensorless Homing but when I home X or Y, Z move...

I wonder what I'm doing wrong?

Here is the Sensorless Homing code sequence :

#define SENSORLESS_HOMING // StallGuard capable drivers only

  /**
   * Use StallGuard2 to probe the bed with the nozzle.
   *
   * CAUTION: This could cause damage to machines that use a lead screw or threaded rod
   *          to move the Z axis. Take extreme care when attempting to enable this feature.
   */
  //#define SENSORLESS_PROBING // StallGuard capable drivers only

  #if EITHER(SENSORLESS_HOMING, SENSORLESS_PROBING)
    // TMC2209: 0...255. TMC2130: -64...63
    #define X_STALL_SENSITIVITY  8
    #define X2_STALL_SENSITIVITY X_STALL_SENSITIVITY
    #define Y_STALL_SENSITIVITY  8
    //#define Z_STALL_SENSITIVITY  8
    #define SPI_ENDSTOPS              // TMC2130 only
    //#define IMPROVE_HOMING_RELIABILITY
  #endif

BrianS2K, 2020-03-10:
I found the reason of my first issue

But I have allways the second issue (I have others but that'll for after solving this one

janoc, 2020-03-10:
I had the same problem - it is the TMC2130 driver issue. Disable the StealthChop (silent) mode on the Z axis in Marlin and the issue will be gone. 

My hypothesis is that the driver expects certain impedance of the motor to be connected for the StealhChop mode to work correctly and when you connect two motors to the same driver this isn't satisfied anymore. And the driver goes nuts ... The classic Allegro A4988 is "dumb" and doesn't have this issue.

Alternatively you can configure the board to run one motor per driver (the dual Z axis setup), however this tends to desynchronize the Z axis motors fairly quickly - again thanks to the TMC2130 step interpolation when using the StealthChop mode. 

These Trinamic drivers have unfortunately quite a few problems, it is nothing to do with the SKR Pro board.

BrianS2K, 2020-03-11:
I tried, I didn't change anything but maybe I missed something 
I comment :
//#define STEALTHCHOP_Z

Tomorrow I'll try with a A4988 and cross my fingers...

But I saw also this line as I have TMC2130 V3.0 SPI should I also uncomment this ? : 
  /**
   * Software enable
   *
   * Use for drivers that do not use a dedicated enable pin, but rather handle the same
   * function through a communication line such as SPI or UART.
   */
  //#define SOFTWARE_DRIVER_ENABLE

janoc, 2020-03-13:
My configuration is here:
https://github.com/janoc/BIGTREETECH-SKR-PRO-V1.1/tree/mendel90/firmware/Marlin-bugfix-2.0.x%20SKR%20ProV1.1-Dmeo/Marlin-2.0.xSKR%20ProV1.1-Demo/Marlin 

Out of top of my head things that could cause issues:
- make sure you have 800mA current selected (in Configuration_adv.h)
- **do make sure you have heatsinks on the drivers** - super important because these drivers get really hot and throttle the motor current otherwise
- disable StealthChop as you did. I have it turned off on all axes and the extruder too.
- disable StallGuard and sensorless homing to make sure it is not getting triggered.
- enable MONITOR_DRIVER_STATUS and TMC_DEBUG. Check the messages/status in the serial console. If the driver is overheating or has some problem, you will see it there.

>But I saw also this line as I have TMC2130 V3.0 SPI should I also uncomment this ? :
No. I don't have it enabled neither. If that was required your motors wouldn't do anything at all because the drivers would never get enabled.

BrianS2K, 2020-03-15:
Hello,
I followed your advice, disabled everything, swaped drivers and I have always the issue.
When I use Single Z motor (Jumpers on the second Z port)

[SKR PRO V1.1 Single Z.mp4.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/files/4334019/SKR.PRO.V1.1.Single.Z.mp4.zip)

When I use Dual Z motor

[SKR PRO V1.1 Dual Z.mp4.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/files/4334018/SKR.PRO.V1.1.Dual.Z.mp4.zip)

janoc, 2020-03-15:
Uh, not sure what did you record those videos with because I had to use 3 different video players to play them - one was showing only a black screen and had sound, another showed only a static image with sound and third one showed video but no sound ...

That noise that you have when you connect two motors to a single driver is exactly the same what I had. For me disabling the StealthChop and setting the current to 800mA (also check the trimpot on the driver itself!) has resolved the problem.  

Make sure  you don't attempt to run too high microstepping neither - e.g. I am at stock 16 microsteps and that you have followed the troubleshooting section here:

https://marlinfw.org/docs/hardware/tmc_drivers.html

Other things to try are replacing the steppers with something that has smaller inductance and/or increasing the operating voltage - e.g. Prusa runs these at almost 28V instead 12. However, that's an expensive and rather invasive solution, depending on how your machine is built. 

If that doesn't fix it, then you have basically two options - either give a separate driver to each motor (the board has support for that) or swap the driver for something else, e.g. the A4988. This problem is unrelated to the SKR Pro.

BrianS2K, 2020-03-19:
Hello,

I tried with A4988 and the problem was the same so went back to the TM2130...
But I found a partial solution of the problem :
I checked the Z estep and they were at 4000 => changed to 400 and it turns.
unfortunatly whether I do UP or Down it only goes up ????
And I also have a horrible noise as long as the stepper is not automatically deactivated after 2 minutes ...

I start thinking possible that it is a mainboard issue :-(

janoc, 2020-03-20:
Ah that Z estep at 4000 is waaay too high. Normal 200 step/rotation motor and 16 microsteps has 200*16 = 3200 for a leadscrew with 1mm pitch. With my leadscrews that have 2mm pitch but 8mm lead, I have only 400 (200*16/8 = 400). So yes, that could well cause a problem because the motor is simply unable to keep up with the changing current from the driver. 

The interesting noises the motors make when idle (it sounds like "psssshhh...shhh") are due to the Trinamic drivers. I was surprised by it too. There are some videos on Youtube on how to tune the parameters of the spread cycle mode so that it is not so noisy. 

The motor noise and the torque issue has zero to do with the motherboard because there is nothing on it that has any impact on that. That's all motor drivers & Marlin setup. 

The rotation direction problem is also most likely due to your Marlin build. But check whether the DIR pin on the driver sockets isn't shorted to something else by accident and that you have put the jumpers configuring the board for the different drivers correctly, especially if you were swapping things around for the Allegro driver.

BrianS2K, 2020-03-22:
Hello, 
Thanks for all those informations and you're noise description is right :-D
I din't put the jumpers on the motherboard, it cames factory configured for 6x TMC2130.
I checked the pins and they are OK
Where in Marlin can I found the direction configuration ?

janoc, 2020-03-22:
> I din't put the jumpers on the motherboard, it cames factory configured for 6x TMC2130.

Hmm, then I do wonder. The documentation shows clearly that for testing with the Allegro driver you need to put in a jumper and for configuring the TMC2130 in SPI mode, you need to put 4 jumpers per motor in a different place and remove that one for the Allegro driver.

Are you really sure you have it set up correctly? My board came preconfigured for the TMC2130 in SPI mode, with the jumpers in place because I have ordered the drivers with it. But I haven't been testing different drivers in it. 

See the manual here, page 7: https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/blob/master/manual/SKR%20PRO%20V1.1%20user%20manual%20v1.1.pdf 

If these jumpers are wrong, then all bets are off. 

The STEP/DIR pins are defined in:
Marlin/src/pins/stm32/pins_BTT_SKR_PRO_V1_1.h

However, I don't recommend messing with that, there is no reason for that to be wrong. You can always put a multimeter/oscilloscope probe on the DIR pin and see whether the value changes when you change direction of movement.

Charly333, 2020-04-04:
Hi, had the same issues when i first swapped Z roods to Pitch 2mm Lead 2mm.  Had to have M203 below Z10  and with M92 Z1616  - on MKS, SKR 1.3 and SKR 1.0 Pro. 

I´ve now switched to Pitch 1mm Lead 1mm and have to have M203 below Z3 and M92 Z3232 . In my start gcode before UBL Bed leveling I reduce to Z2 and raise after leveling back to Z3. This gives me a  Standard Deviation: 0.002713 mm on M48 Z-Probe repeatability Test.