bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #115: How to flash bootloader?

chrstrvs, 2020-03-11:
Hi!

I accidentally shorted a IO pin to 5v and fried the processor. The rest of the board appears to be working, so I'm thinking of swapping the processor for a new one, but I need to flash the bootloader to it.
I understand that I need a ST Link programmer, but how do I hook it up to the SKR Pro board?

Thanks in advance!

Regards,
Christian

GadgetAngel, 2020-03-11:
@chrstrvs Check out this https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/issues/72

chrstrvs, 2020-03-11:
> 
> 
> @chrstrvs Check out this #72

Thanks, that's exactly what I needed!

danandmary, 2020-03-12:
For clarification :
1) Jumper BOOT0 Pin to 3.3V
2) Get and FTDI Adapter, 
3) Download https://www.st.com/en/development-tools/flasher-stm32.html
4) Make sure logic voltage switch on FTDI Adapter is 3.3v
5) Connect to SKR Pro 1.1 as follows Connect 3.3v VCC and ground to UART VCC and ground pins
6) Connect FTDI TX to TFT Connector RX1 and FTDI RX to TFT Connector RX1 (For me the UART TX and RX Pins didnt work)
7) Open Demonstrator GUI, select the correct com port click next.  If error appears, check wires
8) If OK, message will confirm connection Click Next again and and Next Again 
9) In the next screen click the ... near the downoad to device section and select the .BIN file to upload.
10) Click next and wait. 
11) When done, dont forget to remove the BOOT0 jumper.

FYI Marlin firmware does not contain the bootloader, you'll need a copy of firmware which includes the bootloader.

Full Firmware including bootloader : http://cultivate3d.com/pub/media/BTTSKRPro1.bin

chrstrvs, 2020-03-12:
I will probably be using a ST-LINK adapter. Am I right in assuming that I will replace steps 4 to 10 with connecting the ST-LINK adapter to 3.3v, ground and SWD pin and use ST-LINK sw to burn the bootloader?

danandmary, 2020-03-13:
Im sorry I don't have an ST-Link.

npmomchev, 2020-04-03:
Is it possible for someone to share only  **bootloader.bin**  which does not include firmware in self

Thank You

codmpm, 2020-06-01:
As I've obviosly bricked my SKR 1.4, can someone point me to the *bootloader.bin* for it? It's an LPC1769 thus the SKR 1.1 bootloader won't work.

Also its difficult for me to find BOOT0 on the 1.4 board. 
When I understood it correctly it should be R28 according to the schematics: https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/raw/master/BTT%20SKR%20V1.4/Hardware/BTT%20SKR%20V1.4-SCH.pdf
So I have to desolder R28 and pull it's pin low to be able to flash the bootloader.

AnHardt, 2020-06-01:
Dear @codmpm .
Here we are at the BIGTREETECH-SKR-PRO-V1.1 issue tracker. It's using a STM32F407 processor.
LPC-processors are completely out of scope here.

codmpm, 2020-06-01:
Oh, I'm sorry. I did see this when writing my comment.
Thank you!

AnHardt, 2020-06-01:
> I did see this when writing my comment.

And why did you do it anyway?

codmpm, 2020-06-01:
Sorry... *did not* see it.

schnello, 2020-06-20:
> 4. Make sure logic voltage switch on FTDI Adapter is 3.3v

Is this realy 3.3V? 
[image]

c3D-Dan, 2020-06-20:
Yes. Just because there is a Lm7805 offering 5v pins does not mean the logic voltage runs at 5v.  Look at the stm32f4v7 datasheet.

schnello, 2020-06-20:
@c3D-Dan

jeffeb3, 2020-07-04:
@c3D-Dan, Thank you so much for these instructions and .bin file. I managed to bring an skr pro back from the dead, and finding these instructions were key for me. I couldn't get my FTDI dongle to work in windows (I'm not a windows expert), but I did manage to do it in Linux. Here's what I did:

1. Set the Boot0 to 3.3V jumper (it is labeled on the underside)
2. Connect the FTDI Rx to Tx3 (LCD), FTDI Tx to Rx3 (LCD), FTDI GND to GND (I used the LCD pin).
3. Connect the FTDI 3.3V to the 3.3V port near the LCD connector.
4. Installed `stm32flash` (I'm using ubuntu 18.04):
```
sudo apt install stm32flash
```
5. Run the stm32flash with the .bin file you provided, and the port that the FTDI uses:
```
stm32flash -w BTTSKRPro1.bin /dev/ttyUSB0
```
6. It took about a minute, and it confirmed it flashed to 100%
7. Remove the FTDI from the computer, and then undo steps 1 and 2. Don't forget the jumper.

[image]

Thank you again.

qweritos, 2021-10-01:
@c3D-Dan could you please re-upload the bootloader image and/or update the link?

schnello, 2021-10-01:
@qweritos 
https://nextcloud.service-uplink.de/s/ZnYnXwMCPZjrRPL

schnello, 2021-10-13:
@ChrisS1986 
I would do it as descriped it here:
https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/issues/115#issuecomment-653814967

GadgetAngel, 2021-10-13:
> I Do it as descriptet only on Windows. But it dont work, No response from the target, the Boot loader can not be startet.
> 
> Should i remove the wiring and driver for this?
> 
> So i try it tomorrow with ST Link, i ordert a BTT Octopus to replace the Skr 1.1 Pro.

I just got my Octopus Pro board in and I own several Octopus V1.1 board too.  All I do to back up the SD Card bootloder is I use the Original (not a clone) ST-LINK-V2 device and hook it to the SWD connector on the Octopus V1.0/V1.1 board.  Then I bring up https://www.st.com/en/development-tools/flasher-stm32.html Software and hit connect to TARGET.  This always works but if you use a clone ST-LINK-V2.  It may not work.

IMHO, you need to spend the money and purchase an ORIGINAL ST-LINK-V2 if you own any Bigtreetech boards. You can find and ORIGINAL ST-LINK_V2 at https://www.amazon.com/ST-LINK-V2-circuit-debugger-programmer/dp/B00SDTEM1I

The ORIGINAL ST-LINK-V2 might cost $50 but it is like any other tool you buy.  You have to have the **right tool to get the job done**.  I have spent $70 (7 clone ST-LINK_V2 @ $10.00 each) on the clones thinking they died on me.  I wish I would had just laid down the $50.00 for the ORIGINAL at the very beginning!  The clones will NOT connect to some Bigtreetech boards if you are using WINDOWS OS (If you are using LINUX that is a different story)!  The ORIGINAL ST-LINK-V2 under WINDOWS OS will always connect to the board.

vwphan, 2022-01-18:
> @c3D-Dan, Thank you so much for these instructions and .bin file. I managed to bring an skr pro back from the dead, and finding these instructions were key for me. I couldn't get my FTDI dongle to [...]

This method still worked as of 1/17/2022.  I was re-flashing using Ubuntu 20.04, it took several unsuccessful attempts to get the board to flash.  I had to hold the re-set button for approximately 30 seconds, wait a few seconds, then attempt to flash the board.  It took ~5 minutes to flash.

Good as new.  Thanks a bunch!!

pcipri76, 2023-10-27:
@c3D-Dan could you please re-upload the bootloader image and/or update the link?

qweritos, 2023-10-27:
@pcipri76 https://drive.google.com/file/d/1QW1IjdsDbjqhVf-OXsXqE0a_9XmiENuj/view?usp=sharing

pcipri76, 2023-10-27:
> @pcipri76 https://drive.google.com/file/d/1QW1IjdsDbjqhVf-OXsXqE0a_9XmiENuj/view?usp=sharing

Thank you very much

pullasuti, 2024-08-28:
I've been trying to get this to work, but just can't get stm32flash or Demonstrator to talk to the board. I'm thinking my board is truly dead... if I don't jumper BOOT0, the CPU gets very hot, also no memory card LED, and all the stepper LEDs are dimly lit. When jumpering either BOOT0 or BOOT1, the board does show up as a TTY device under Linux, but that's it. 

I've tried it with a USB-TTL adapter and Raspberry Pi GPIO.

Any ideas or just give up and bin it? :)

qweritos, 2024-09-03:
@pullasuti, with a high probability you burned the stm chip, and dimmed LEDs is also a clue. Check 1117 LDO chip, it also gets insanely hot. However, I burned stm many times, but after reflowing&reflashing it always gets back in operation :)