bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #116: Part Cooling Fan not shared between multiple extruders

Rhage, 2020-03-12:
I have a multi-extruder setup with a single nozzle

The Slicer (Cura 4.5) is set to turn the print cooling fan on starting at layer 2.
When layer 2 begins the third extruder is active and the part cooling fan correctly turns on.

However, as soon as the active extruder changes, to either extruder 0 or 1, the cooling fan turns off and only resumes when the third extruder (number 2) becomes the active extruder again.

This continues for the rest of the print. The print cooling fan turns on only for Extruder 2 and is off for extruder 0 and 1

Any help would be appreciated.

Screenshots of Slicer settings and my Marlin Configuration.h and Configuration_adv.h files are attached

![Slicer_Cooling](https://user-
[Configuration .txt](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/files/4321897/Configuration.txt)
[Configuration_adv.txt](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/files/4321898/Configuration_adv.txt)

images.githubusercontent.com/14357200/76479182-fc960f80-63d8-11ea-96ea-e66d4ddcdd1d.PNG)
[image]
[image]
[image]
[image]

oldman4U, 2020-06-08:
Hello Rhage.

Have you been able to solve the reported issue in the meantime?

Please let us know

Rhage, 2020-06-08:
Yes, I was. But not in a way I'd probably advise duplicating.

It appears to be a bug/feature (depending on who you ask) of the Marlin 2.0 implementation of the m106 gcode

In my opinion it's a bug. The specification for that command states that when no fan parameter is provided with m106, which is how Cura sends this command with the setup I showed in my screenshots, it is supposed to default to Fan 0.

However, Marlin is defaulting the command with no fan parameter to the currently active tool.

I solved this by editing the Marlin source to always set the fan speed for m106 to all tool fans, which means my printer is no longer running the standard released version of Marlin 2.

oldman4U, 2020-06-09:
Good to hear you have been able to solve the issue.

Would you please be so kind to close this ticket then, by clicking on the Close button right besides the Comment button..

Thank you

closed 2020-06-09 by Rhage