bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #117: SKR PRO 1.1 + TMC5160, Homing is "working" but no other movement

mcmuff91, 2020-03-22:
Hello everyone,
atm I´am working on a big selfmade Delta-printer for Ceramics.
The setup:
- Skr Pro
- TMC 5160 V1.2 BTT (removed the CLK Pin)
- Nema 23, 3 A, 1,8°
- 24 Volt Power supply
- RepRap Full Graphic LCD

The main problems:

- When i send G28 via Pronterface or via the RepRap-Controller the printer starts to home. But one after the other (x,y,z). There is no movement of all three Axis at the same time, so I have to move the Effector to the upper end of the printer. The axes are hitting the endstops and doing a rebump . But after this there is no movement so the safe zone (enabled DELTA_HOME_TO_SAFE_ZONE)

After homing the Printer is not responding to movement commands via Pronterface or the Controller.

Tried multiple Versions of Marlin ( the BTT-Version, 2.04, 2.05). 
I have no idea how to fix the Problem.

oldman4U, 2020-06-08:
Hello mcmuff91.

Could you please let us know the status of the reported issue and close this ticket, in case the issue has been solved in the meantime.

Thank you

oldman4U, 2020-10-10:
Please help the community and close the ticket once you do not need it anymore. Thank you

evandene, 2020-11-25:
> Hello everyone,
> atm I´am working on a big selfmade Delta-printer for Ceramics.
> The setup:
> 
> * Skr Pro
> * TMC 5160 V1.2 BTT (removed the CLK Pin)
> * Nema 23, 3 A, 1,8°
> * 24 Volt Power supply
> * RepRap Full Graphic LCD
> 
> The main problems:
> 
> * When i send G28 via Pronterface or via the RepRap-Controller the printer starts to home. But one after the other (x,y,z). There is no movement of all three Axis at the same time, so I have to move the Effector to the upper end of the printer. The axes are hitting the endstops and doing a rebump . But after this there is no movement to the safe zone (enabled DELTA_HOME_TO_SAFE_ZONE)
> 
> After homing the Printer is not responding to movement commands via Pronterface or the Controller.
> 
> Tried multiple Versions of Marlin ( the BTT-Version, 2.04, 2.05).
> I have no idea how to fix the Problem.
> 
> [Marlin config 2.0.5.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/files/4365559/Marlin.config.2.0.5.zip)

Did you get your DELTA Printer to work?