bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #118: Serial ports are gone

Charly333, 2020-03-28:
Had SKR PRO V1.1 working with 6x 5160 for a couple of month. Now after a restart TFT 35 and Octoprint won´t connect with serial ports any more.

Reflashed firmware that did run before: No serial found, neither by TFT 35 nor Octoprint or PC.
Flashing works, green LED flickers on start and firmware.bin is renamed to FIRMWARE.CUR. Did test several reformatted micro SD cards. Any LED shows the state expected.

First try: ---------------------------------------------------------------------

Reflashed BTT Marlin demo version with 
#define SERIAL_PORT -1
#define SERIAL_PORT_2 1 
#define  BAUDRATE 115200

The board is found in general. Windows 10 device manager prompts an "STMicroelectronics Virtual Com Port (COM5)" - but pronterface hangs whenever I try to connect the board via Com 5 and prompts this error lines .

Connecting...
[ERROR] Could not connect to COM5 at baudrate 115200:
Serial error: Cannot configure port, something went wrong. Original message: WindowsError(433

Windows Device manager on TMicroelectronics Virtual Com Port prompts: "status - working, no problems" but also knows in events - devide not migrated / still some installations needed.

Even if STM webside states that Win 10 doesn´t need any additional drivers, I installed the BTT STM Drivers but same results.

Second Try ---------------------------------------------------

Adapted newest original Marlin 2.0 with same Serial-Port Settings. After reflashing that firmware, board is not found as with old - former running - Version.

------------------------------------------------

Any idea hot to get it back running?

Reg. Charly

c3D-Dan, 2020-04-14:
Time to replace your stm32f4 cpu or your board.  If you do replace your CPU, you’ll need to initially upload firmware via FTDI adapter.  Details here https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/issues/115

dryflow, 2020-05-02:
Had the same behavior, SKR PRO with TMC2209, BL-touch, sensorless homing on XY, fans over EXP2, eeprom on SD, worked several weeks like a charm. Two days ago prints gone bad after a few layers and I wondered why. Thought it could be the nozzle, so I changed it. After this procedure the printer made a few moves, heated up and then nothing else. TFT and Octoprint lost connection.

Make a long story short: The termistor at the hotend had a problem, caused a short circuit. After I changed it and put the Marlin2.0 back everything working like a charm now. No need for replacing the board or the cpu. Maybe worth a try to check cabling and electric components.

oldman4U, 2020-06-08:
Hello Charly333.

Could you please let us know the status of the reported issue. In case the problem has been fixed in the meantime, it would be great if you could close your ticket.

Thank you

Charly333, 2020-06-11:
Had different experiences. Sometimes new Card and reflashing the board brings it back to live - sometimes - on other boards - it doesn´t. Specially the thermistor channels in SKR boards will crash the board if they are getting in contact with some voltage from the heater or so.

closed 2020-06-11 by Charly333