bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #119: No firmware update possible via SD card

chrisdruck, 2020-03-30:
I tried to make a setting in Marlin 2.0 bugfix so that I can describe the epromm and then I loaded the Firmware.bin. Because I could not set a command M500 Now I can no longer load firmware via SD card. Can I delete the firmware from the board. I can run M997 but then no M500

kolbolt, 2020-04-01:
I have been trying to fix or get the BIG-TREE-TECH skr pro V1.1 to work with marlin 2.0x bugfix by following this video on youtube [](https://www.youtube.com/watch?v=VsKHwYtqLz0).  

And I got this for an error. 

Processing BIGTREE_SKR_PRO (platform: ststm32; board: BigTree_SKR_Pro; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/BigTree_SKR_Pro.html
PLATFORM: ST STM32 6.0.0 > STM32F407ZG (192k RAM. 1024k Flash)
HARDWARE: STM32F407ZGT6 168MHz, 192KB RAM, 1MB Flash
DEBUG: Current (stlink) External (jlink, stlink)

PACKAGES:
 - framework-arduinoststm32 3.10700.191028 (1.7.0)
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 17 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <LiquidCrystal> 1.5.0
|   |-- <Wire> 1.0
|-- <TMCStepper> 0.6.2
|   |-- <SPI> 1.0
|-- <Adafruit NeoPixel> 1.3.5
|-- <U8glib-HAL> 0.4.1
|   |-- <Wire> 1.0
|   |-- <SPI> 1.0
|-- <Adafruit MAX31865 library> 1.0.3
|   |-- <SPI> 1.0
|-- <LiquidTWI2>
|   |-- <Wire> 1.0
|-- <Arduino-L6470> 0.8.0
|-- <SailfishLCD>
|-- <SailfishRGB_LED>
|   |-- <Wire> 1.0
|-- <SlowSoftI2CMaster>
|-- <SPI> 1.0
|-- <Servo> 1.1.2
|-- <IWatchdog> 1.0.0
|-- <Wire> 1.0
Building in release mode
Compiling .pio\build\BIGTREE_SKR_PRO\src\src\HAL\STM32\HAL.cpp.o
Compiling .pio\build\BIGTREE_SKR_PRO\src\src\HAL\STM32\HAL_SPI.cpp.o
Compiling .pio\build\BIGTREE_SKR_PRO\src\src\HAL\STM32\Sd2Card_sdio_stm32duino.cpp.o
Compiling .pio\build\BIGTREE_SKR_PRO\src\src\HAL\STM32\Servo.cpp.o
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:32:0,
                 from Marlin\src\HAL\STM32\timers.h:25,
                 from Marlin\src\HAL\STM32\HAL.h:121,
                 from Marlin\src\HAL\STM32\HAL.cpp:26:
Marlin\src\HAL\STM32\../../inc/../pins/pins.h:652:6: error: #error "BOARD_BIGTREE_SKR_PRO_V1_1 has been renamed BOARD_BTT_SKR_PRO_
V1_1. Please update your configuration."
     #error "BOARD_BIGTREE_SKR_PRO_V1_1 has been renamed BOARD_BTT_SKR_PRO_V1_1. Please update your configuration."
      ^~~~~
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:32:0,
                 from Marlin\src\HAL\STM32\HAL_SPI.cpp:25:
Marlin\src\HAL\STM32\../../inc/../pins/pins.h:652:6: error: #error "BOARD_BIGTREE_SKR_PRO_V1_1 has been renamed BOARD_BTT_SKR_PRO_
V1_1. Please update your configuration."
     #error "BOARD_BIGTREE_SKR_PRO_V1_1 has been renamed BOARD_BTT_SKR_PRO_V1_1. Please update your configuration."
      ^~~~~
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:40:0,
                 from Marlin\src\HAL\STM32\timers.h:25,
                 from Marlin\src\HAL\STM32\HAL.h:121,
                 from Marlin\src\HAL\STM32\HAL.cpp:26:
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:1553:4: error: #error "HEATER_0_PIN not defined for this board."
   #error "HEATER_0_PIN not defined for this board."
    ^~~~~
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:1751:8: error: #error "E0_STEP_PIN, E0_DIR_PIN, or E0_ENABLE_PIN not defined for this
 board."
[... 81 lines cut ...]
mks_robin                 IGNORED
mks_robin_pro             IGNORED
mks_robin_lite            IGNORED
mks_robin_lite3           IGNORED
jgaurora_a5s_a1           IGNORED
STM32F103CB_malyan        IGNORED
chitu_f103                IGNORED
STM32F401VE_STEVAL        IGNORED
FLYF407ZG                 IGNORED
FYSETC_S6                 IGNORED
STM32F407VE_black         IGNORED
BIGTREE_SKR_PRO           FAILED    00:00:12.986
BIGTREE_GTR_V1_0          IGNORED
BIGTREE_BTT002            IGNORED
teensy31                  IGNORED
teensy35                  IGNORED
esp32                     IGNORED
linux_native              IGNORED
SAMD51_grandcentral_m4    IGNORED
rumba32_f446ve            IGNORED
rumba32_mks               IGNORED
include_tree              IGNORED
==================== 1 failed, 0 succeeded in 00:00:12.986====================

can anyone help me out

oldman4U, 2020-04-01:
Hello Kolbolt.

Would be great if you could start your own ticket instead of using the ticket of another user.

You can find the error in the text you pasted above, where it says:
#error "BOARD_BIGTREE_SKR_PRO_V1_1 has been renamed BOARD_BTT_SKR_PRO_V1_1. Please update your configuration."

So please change the platform.ini file accordingly which should solve this problem.

Hope this helps

oldman4U, 2020-04-01:
@chrisdruck 

Sorry that i can not help with your problem

kolbolt, 2020-04-01:
> @chrisdruck
> 
> Sorry that i can not help with your problem

Ok

LegendAlexander, 2020-04-09:
you may have done this but did you move the jumper over to usb before starting the motherboard

LegendAlexander, 2020-04-09:
then just clear the sd and add the new firmware

oldman4U, 2020-06-08:
Hi chrisdruck.

Seems this ticket has not been used for a while, so could you please close it.

Thank you