bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #121: sensorless probing : not compiling ...

davidmarli, 2020-04-05:
Hi,

Itry to test sensorless probing, but marlin do not want to compile : 
Skr pro 1.1 + 2130 spi divers. Marlin 2.0.5
I uncomment     #define SENSORLESS_PROBING
When I try to compil : 

#error "SENSORLESS_PROBING requires a TMC stepper driver with StallGuard on Z."

How to fix it ? What am I wrong doing ?

Thank you for your help

Mariu86, 2020-04-06:
You must set the extruder 0 with tmc driver.

davidmarli, 2020-04-07:
Hi,

alreadey done, I have 5 tmc2130 driver.
In marlin, I wrote TMC2130 for driver X,Y, Z E0 and E1.
Still same bug when compiling.

oldman4U, 2020-06-11:
Hello davidmarli.

Could you please let us know the status of this issue? Please attach your config and config.adv file, in case you could not solve it in the meantime.

Thank you

davidmarli, 2020-06-12:
Work

closed 2020-06-12 by davidmarli