bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #122: SKR Pro V1.1 Question about two Fans

MikeVanDike, 2020-04-05:
Hi Bigtreetech-Team,
i have a Question for you.
I want to cool my printet Object with two Fans.
How can i realize this?
Is it possible to connect two Fans on one Output (Fan0)? (whats the max Amp fore one Fanoutput?)
I try this and it smels not good :-S 
Or can i conncet one Fan on Fan0 and another one on Fan1 and syncronice it in the firmware? 
But how can i do this? :-/

LegendAlexander, 2020-04-09:
You may find this video helpful
[https://youtu.be/vweMI8OHxmY](url)

MikeVanDike, 2020-04-15:
I know Edwards Channel and this Video, but it is not helpful for me :-/
How can i define that when Fan0 is on that Fan1 is also on and Fan1 goes every time the same Fan Speed as Fan0?

BunniKiller, 2020-05-12:
you  could run the fans in a series wiring, but they will run at 1/2 speed...
POS volt supply---red wire---fan1---black wire----|---red wire---fan2----black wire----NEG volt supply..

GhostlyCrowd, 2020-05-12:
> you could run the fans in a series wiring, but they will run at 1/2 speed...
> POS volt supply---red wire---fan1---black wire----|---red wire---fan2----black wire----NEG volt supply..

this is a bad idea unless both fans have the EXACT impenitence one will run and the other will most likely spin at all. 

The correct way to do this is to use the CORRECT settings in marlin to opwerate extra fans, Look at the Auto Extruder fan configs in the configuration_adv.h  and also  Controller Fan

```#define USE_CONTROLLER_FAN 

//#define E0_AUTO_FAN_PIN -1
//#define E1_AUTO_FAN_PIN PE6 //Stepper cooling, turns on with Exturder
//#define E2_AUTO_FAN_PIN -1
//#define E3_AUTO_FAN_PIN -1
//#define E4_AUTO_FAN_PIN -1
//#define E5_AUTO_FAN_PIN -1
//#define E6_AUTO_FAN_PIN -1
//#define E7_AUTO_FAN_PIN -1
//#define CHAMBER_AUTO_FAN_PIN -1

#define EXTRUDER_AUTO_FAN_TEMPERATURE 50
#define EXTRUDER_AUTO_FAN_SPEED 255   // 255 == full speed
#define CHAMBER_AUTO_FAN_TEMPERATURE 30
#define CHAMBER_AUTO_FAN_SPEED 255```

BunniKiller, 2020-05-12:
if one has the file to open and look at it and do the proper "repairs" would be kewl, but sometimes all we are given is the firmware.bin file and I have yet to find a way to open and modify a bin file.  Im really new at this and have way more to learn than I have time... literally... cancer doesnt care about how much time one needs... but Im gonna learn as much as I can :)

MikeVanDike, 2020-05-26:
@GhostlyCrowd 
Is it right when i make this

//#define E0_AUTO_FAN_PIN PC8 & PE5
or how can i synchronice it?

BunniKiller, 2020-05-26:
now thats a way I didnt think about..  but it works in ladder logic and
basic, I dont see why it wouldnt work in C++ type of language. Im not sure
of the "and" in C++ is... ( if its an ampersand or the word "and" or a +
sign)
give it a try, if it doesnt work then all thats been lost is some time.
seriously dont believe it will do any physical damage to the hardware.