bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #123: SKR Pro V1.1 stops while working with octoprint

samgozman, 2020-04-08:
Serial port falls off during printing through an octopoint.
[Detailed video](https://youtu.be/rcAsfghPEvI)
Tried stock firmware and marlin 2.0.5. Octoprint connection is working only with baudrate  115200, and drops while printing.

To others, the problem with this board is:
- sd card never worked on it (No media) (sdsupport on)
- USB never worked 

Also, the BigTreeTech TFT35 v3.0 display can never read the SD card and USB (read usb error).
Which leads me to the idea that I don’t have a single way to communicate with the SKR pro.

MikeEitel, 2020-04-21:
I agree that the impossibility to write a new .bin file via usb is a show stopper. Pitty me found that after i bought two complete mcu, lcd, driver combo's. Nice hw but not good supported sw. A shame.
Mike

phlebaspho, 2020-05-26:
In my case, after much A/B testing I realised it was not the TFT35 - or Pronterface it was "device connected to Marlin serial port 2" that experienced the the comms loss. Moving the TFT35 to the first serial port in Marlin config.h fixed it.

I don't have an SDcard issue though - so I can reflash firmware at will - so this might not help you.

oldman4U, 2020-06-09:
Hello samgozman.

Have you been able to solve the reported issues in the meantime?

Please let us know

Thank you

samgozman, 2020-06-10:
@oldman4U The problem is partially solved: SD cards working, but serial port via octoprint or simplify3d is dropping no matter the cable. And USB AM is still a dumb stub without any functionality

oldman4U, 2020-06-11:
Hi.

Great to hear the SD card is working now. Is this the SD card from the board or the TFT?

Yes, the missing USB drive support is a shame and many users would like to see this functionality.

Unfortunately i can not help with Octoprint - i am not using it.

oldman4U, 2020-06-11:
.... and have you tried to change the serial ports, like described here?

https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/issues/112

samgozman, 2020-06-14:
Yes, I studied all issues before publish mine. At first, the motherboard didn’t work on the wire at all, then I set it up so that it would at least respond to commands. But the connection falls off when printing in the first minutes after warming up. I tried several different cables, Raspberry pi Zero, Orange Pi Zero, Windows PC, MacOs - no luck.
I tried several combinations of serial ports. On some, the display did not work properly. Now I settled on the following combination:
```
#define SERIAL_PORT 2
#define SERIAL_PORT_2 -1
```

oldman4U, 2020-06-15:
Hi.

Changing the serial port solved the problem. I expect you have see and activated the Octoprint related setting in the TFT firmware:

Keep Serial always On (ONLY SUPPORTED ON TFT24 V1.1, TFT35 V3.0, AND TFT28 V3.0)
 
Keep UART (Serial communication) alive in Marlin Mode
Allow seamless OctoPrint UART connection to the TFT's UART/serial expansion port no matter which mode the TFT is in.

Options:  0: Disabled    1: Enable
#define SERIAL_ALWAYS_ON 0  // Default: 0 (Disabled)