bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #125: How Do You Print On A SKR PRO

LegendAlexander, 2020-04-09:
I have seen many comments and posts of the SD card slot not working and the USB port not working. I just want to know how can i print something on this board without having to buy some thing.

First let me say I am not an expert in any of this in fact I am nearly average so when you explain something please make it as simple as possible. My goal with this post is to learn how to print WITH THIS BOARD. Currently all I can do is extrude filament. Amazingly though I got everything else working thanks to the help of this community! I just want to print normal G-code from CURA. Before this main board on the stock ender 3 i would just insert an SD and navigate the lcd and print…done. On this one when I click “Init. Media” nothing happens.  

My Setup:

- Standard Ender 3 hardware including display
- Skr pro v1.1 main board
- added another Z stepper motor ( So in total I have 2 Z stepper motors )
- 4x TMC2209 ( UART MODE )
- Marlin 2.0.3 For SKR PRO [https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/tree/master/firmware/Marlin-bugfix-2.0.x%20SKR%20ProV1.1-Dmeo/Marlin-2.0.xSKR%20ProV1.1-Demo
[image]
(url)

Hopefully with all your help I can get this working then ill have a fully functioning printer again! And I would also make a post explaining everything I did to get to this point which I think will help people facing similar problems with this board.

I have access to:

- Atom for Mac
- PronterFace for Mac
- Windows PC
- Mac PC

BunniKiller, 2020-05-12:
try using the bigtreetech_skr_pro_v1.1-master files/folder...
open/extract the zip and look in folder FIRMWARE, then open folder skrprov1.1 - marlinv2.0-12864p-16div-usually, copy the "FIRMWARE.bin" file and put on the SKR Pro board via the SD micro slot... turn on board and let it boot,   Things should work from that point...

and BTW is that a tft 35 video with touchscreen? if it is why are you using the encoder button? The touch screen option is sooo nice.

reipet, 2020-05-30:
I have the same question. I finally got the Board running as I need it and could print through Pronterface already, later will add Octoprint but for now I need the SD-Card Option.

Im using the SKR Pro v1.1. on a Ender5 with Dual-Z and TMC2209, BLTouch and additional Eeprom on the I2C Port. I also have an BTT TFT35 V2.0 but neither SD-Slot is working. I can select both (On-Board and TFT one) from the menu but then its stuck in "Loading"

How you Guys get this Board to print at all? Only via external G-Code stream?

oldman4U, 2020-06-11:
Hello @LegendAlexander.

Could you please let us know the status of the reported issue. Please close this ticket in case the problem has been solved in the meantime.

@reipet. Could you please start a new ticket, so someone can try to help. This ticket seems to be inactive.

Thank you

oldman4U, 2020-10-10:
Please help the community and close the ticket once you do not need it anymore. Thank you

mitchcecg, 2021-01-14:
Did you eve find the answer... I have the same issue ???