bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #126: MKS MINI12864 v2.0 and possibly others

Black6spdZ, 2020-04-12:
not sure how to do the pull request stuff but I had to modify the pins_BTT_SKR_PRO_v1_1.h with this to get the display to work - added just below ULTIPANEL entry:

#if ENABLED(MKS_MINI_12864)
      #define DOGLCD_CS PG3
      #define DOGLCD_A0 PG6
      #define LCD_PINS_RS PD10
      #define BTN_EN1 PG10
      #define BTN_EN2 PF11
      #define SD_DETECT_PIN PF12
      #define LCD_SDSS PB12
      #define LCD_PINS_ENABLE PD11
      #define LCD_PINS_D4 PG2
      #define LCD_CONTRAST_MIN 0
      #define LCD_CONTRAST_MAX 255
      #define DEFAULT_LCD_CONTRAST 180
    #endif

lolitstony, 2020-07-22:
I'm trying to get my mks min12864 to work but I have no idea how to connect it to my SKR mini e3 v2.0
I uncommented #define MKS_MINI_12864 in the Configuration.h and compiled
I connected the screens EXP 1 and EXP 2 to the EXP 1 on the board using jumper cables and I get a bright white screen.
I used the pin diagram on https://reprap.org/wiki/MKS_MINI_12864 but I don't know how to correspond them to the pins on the board. 
I checked the pins_BTT_SKR_MINI_E3_V2_0.h and I'm not sure which pin corresponds to what since the name of the pins are different on the screen diagram

[image]
.

[image]

mihal1951, 2020-12-15:
> не знаю, как делать запросы на вытягивание, но мне пришлось изменить pins_BTT_SKR_PRO_v1_1.h, чтобы заставить дисплей работать - добавлено чуть ниже записи ULTIPANEL:
> 
> #if ВКЛЮЧЕНО (MKS_MINI_12864)
> #define DOGLCD_CS PG3
> #define DOGLCD_A0 PG6
> #define LCD_PINS_RS PD10
> #define BTN_EN1 PG10
> #define BTN_EN2 PF11
> #define SD_DETECT_PIN PF12
> #define LCD_SDSS PB12
> #define LCD_PINS_ENABLE PD11
> #define LCD_PINS_D4 PG2
> #define LCD_CONTRAST_MIN 0
> #define LCD_CONTRAST_MAX 255
> #define DEFAULT_LCD_CONTRAST 180
> #endif

I cannot find most of the lines in the pins_BTT_SKR_PRO_v1_1.h file. If you added this to the file, what did you do with the source files (#elif ENABLED (MKS_MINI_12864)
    #define DOGLCD_A0 PG6
    #define DOGLCD_CS PG3
    #define BTN_EN1 PG10
    #define BTN_EN2 PF11
  #else

    #define LCD_PINS_RS PD10

    #define BTN_EN1 PG10
    #define BTN_EN2 PF11

    #define LCD_PINS_ENABLE PD11
    #define LCD_PINS_D4 PG2)