bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #127: How to add a filament runout sensor on skr pro

LegendAlexander, 2020-04-14:
I cant find any probe pins and i tried adding the sensor to the Xmax but it still wont act knowledge when filament is out. Actually i cant get the Xmax to trigger at all. 

Aside from that i dont know where to remap the filament runout pin to since it seems there isnt one on the board. 

I tried making this #Define Filament_Runout_Pin X_Max_Pin

Any ideas?

SrGeek, 2020-04-20:
plug in an endstops 1st and see if the led light next to the E0 turns on.  If it does the pin works you just have to find the right Marlin changes

AustrianBossFan, 2020-04-21:
i had the same problem today. i'm using a mechanical endstop as filament rounout sensor.
the board has 3,3v on the vcc pins of the endstop connectors. when high the signal pin of the endstop only had 1.5v. although the endstop is labeld as 3-12v it does not work with 3,3v.

for x and y i'm using optical endstops, they work fine with 3,3v.

i ended up using the extension-2 connectors 5V, GND and PE4 pins:
[image]

you need to change the filament runout pin in pins_BTT_SKR_PRO_V1_1.h:

#ifndef FIL_RUNOUT_PIN
  #define FIL_RUNOUT_PIN PE4
#endif

Baltazart, 2020-04-21:
Marlin 2.0.5.3 SKR-Pro v1.1 + Smart Filament Sensor

"pins_BTT_SKR_PRO_V1_1.h
#ifndef FIL_RUNOUT_PIN
#define FIL_RUNOUT_PIN PE15  //"PE15=Pin E0 X_MAX_PIN" SKR-Pro v1.1, SFS Smart Filament Sensor.
#endif
*****
"configuration.h"
#define FILAMENT_RUNOUT_SENSOR
#if ENABLED(FILAMENT_RUNOUT_SENSOR)
#define NUM_RUNOUT_SENSORS   1    
#define FIL_RUNOUT_INVERTING true    //<<< pour moi c'est "true"
#define FIL_RUNOUT_PULLUP          
//#define FIL_RUNOUT_PULLDOWN   
#define FIL_RUNOUT_PIN PE15    //<<<<SKR-Pro-v1.1 stock sensor E0 X_MAX_PIN. 
*****
#define FILAMENT_RUNOUT_SCRIPT "M600"
#define FILAMENT_RUNOUT_DISTANCE_MM 7
#ifdef FILAMENT_RUNOUT_DISTANCE_MM
#define FILAMENT_MOTION_SENSOR
*****
configuration_adv.h
#define LIN_ADVANCE     //j'ai activer pour "Smart Filament Sensor" BTT SKR-Pro v1.1.
*****
Here is an SFS configuration that works for me 100% and we are entitled to a flashing E0_X_MAX LED.

To0wnn, 2020-06-10:
The PE15 I want to use for filament run out but somehow is does not work. It is assigend correctly in pins.h and using using marlin bugfix 2.0.x and with tft35 v3
Any ideas? With the M119 command i see it is working correctly.
[https://to0wnn.com:5001/sharing/EmN4oMFoo](https://to0wnn.com:5001/sharing/EmN4oMFoo) Here are my configs

oldman4U, 2020-06-11:
Hello @LegendAlexander. Have you been able to solve the reported issue? If so, could you please close this ticket, thank you.

@To0wnn: This ticket seems to be inactive for some time now. Could you please open a new ticket so someone can try to solve your problem.

Thanks