bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #128: Weird behavior after plugging stepper motor

mqt0029, 2020-04-15:
I got mine to work on the test bench, so I took the motors drivers out to put on some heatsinks before actually drives the motors, and I plugged one of them in backward (or upside down I guess, doh!). Now the boards won't boot (TFT says no printer attached, command from ESP wifi doesn't reply anything, tried to reflash the firmware but it doesn't load the firmware.bin into FIRMWARE.cur)

Drivers are TMC2130 on XYZ and TMC2209 on E0,1,2, I plugged the X one in backward.

I've also checked for connectivity, in case I blew something, everything is good. I've also tried to reflash using ST-Link V2, but it says read out protected and I cannot disable it.

Any suggestion as to what to do now?

GhostlyCrowd, 2020-04-29:
you've fried the MCU Time for a new board

oldman4U, 2020-06-11:
Hello mqt0029.

Sounds like a hardware issue. Could you please be so kind and Close this ticket by clicking on the Close button right beside the Comment button.

Thank you

closed 2020-06-15 by mqt0029