bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #129: Skr pro tmc 2130 :tmc connection error

davidmarli, 2020-04-16:
Hello.
Just biught thé board. 
I plug à tft3. 5 v3. 0 on it. 
And 5 tmc 2130 spi v3. 0
In marlin I déclare tmc 2130 as driver for x y z e0 and e1. 
Disabled all stealthchop and stallguard to test (if i enabled it that s thé same).
When i switch on, tft connectés to thé printer. Température are OK. I can heat thé bed and extruder, but it says tmc connection error and no motors can be driven... 
Nothing happens... 
Jumpers are OK (i tthink, 4 Jumpers per drivers in spi sockets.) 

Thank you for your help... 

Ps : marlin 2.0.5.3

kkosa1, 2020-04-16:
search online for "gadgetangel" and you will find a huge document.  starting around page 199 for your issues.  I tried to forward it to you but file was too large.  I followed the guide and all is working fine for me for last few months.  Also be sure to enable M122 function in Marlin via     #define MONITOR_DRIVER_STATUS on line 2265 of bugfix from 4-15-20 in configuration_adv.h  as well as   #define TMC_DEBUG at line 2345 also in configuration_adv.h

oldman4U, 2020-06-11:
hello davidmarli.

Great to hear that everything works. Could you please help us to keep the ticket system useful by closing unneeded tickets.

Thank you

closed 2020-06-12 by davidmarli