bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #130: Drv8825 drivers not recieving power

SrGeek, 2020-04-17:
I check vfer to adjust but there is no voltage across the driver, attached is a picture of the pins for MS1,MS2,MS3 and D.  Marlin setting changed drivers to drv8825.  need help thanks
[image]

SrGeek, 2020-04-27:
really guys nothing?

oldman4U, 2020-05-11:
hi.

Have you see this guide made by an angel: https://github.com/GadgetAngel/SKR-PRO-V1.1-Stepper-Driver-Jumper-Configuration-Manual

Hope this helps

SrGeek, 2020-05-16:
@oldman4U Yes that’s why I put the pics.  They match the document.  I haven't seen any one resolve this in any of the forums either.  !
Most give up and just wait on the TMC2209 or something similar.  Pretty disappointed

oldman4U, 2020-05-17:
I had a similar problem with A4988 with green pcb. Finally i switched to
2209 with sensorless homing and i love them. You can only hear the noctua
fans and they are well known to be very very quiet.

Hope they will work for you like they do for me!😁

oldman4U, 2020-06-08:
Hello SrGeek.

What is the status of the issue?

SrGeek, 2020-06-13:
Still not resolved.  I got 2130 and 5160, I still cant get board to work.  Driver bad register problems.

SrGeek, 2020-06-13:
Not working is a stretch.  I can heat the bed, turn on the LCD display, turn on fans...but drivers....silent stepper drivers are too silent because they dont turn on.

SrGeek, 2020-06-15:
I have continuity between the power and ground pin of the motor power, once I changed the 15A fuse.  pretty frustrated.

oldman4U, 2020-06-15:
So a broken fuse caused this issue? And you found the issue - this is great!

SrGeek, 2020-06-18:
Motor power + and - are continuous.  My board is fried on all the driver sockets side.  Other than that everything else works, heaters, LCD etc.  My march order of SKR Pro boards never arrived (sad face) and Amazon doesn't sell them anymore.   So l am keeping the board stored somewhere maybe one day the PCB layout will become available and I will find what component  that got fried so I can solder in a new one.    Closing this issue.

closed 2020-06-18 by SrGeek

oldman4U, 2020-06-18:
Hi. Have tried to change the fuse ?