bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #132: What a Bloody board !!!

davidmarli, 2020-04-24:
Hi, can't get printing with skr pro 1.1 !!!
Always bug.
I tried yet qith marlin 2.0.x bugfix ...
Now nothing is moving : homing failed reset ....

That's really shit ...

What version of marlin do you use and have you managed to uise this board ???

Thnak you and sorry, but that's a week I work on it all day, and always bug ...

GhostlyCrowd, 2020-04-24:
I'm on 2.0.x-Bugfix (bleeding edge) and its working fine for me. 

I'd recommend turning off flash eeprom emulation and using SD though.

davidmarli, 2020-04-24:
Now an another bug : I try to print an object.
And about a few couple of minutes all is shifted ....
That's a CoreXY. Befor having this board I had a Re-arm+ramps, and it never failed !!!

Do you think disables eeprom emulation can do something ???

GhostlyCrowd, 2020-04-24:
if its shifted your steppers or drivers are overheating and missing steps. whats your actual hardware.  I am using this board with 2209's with out an issue and I had to switch to sd eeprom because it was bugged for my BLTOUCH

davidmarli, 2020-04-24:
Dont rhinl so.
Tmc2130 voltage 0.7.
Current max in marlin 1000
There is à 12v fan 80mm at full stock. When you touch thé drivers (after 30 min pronting) They are cold... 

Now I try baudrate 115200 instead of 250000, my last chance...

GhostlyCrowd, 2020-04-24:
While its plausible that you have a defective board, you've gone from "no moving at all" to "now another bug layer shifting"  it sounds more like software or user error related than the board its self's fault.  My 2130's have had ongoing random layer shifting on marlin for over a year to the point that I went with Klipper on that unit and haven't had a layer shift since.

davidmarli, 2020-04-24:
I think I just find the pb : 
I wanted to touch the driver to see how warm they was, and I just touch the motor Y cable and it moves !!! The printer jump steps !!!

My wires are with dupont, and it seems not to connect well on the board !!
What a mess ....

I don't how to improve connnection ...

GhostlyCrowd, 2020-04-24:
> I think I just find the pb :
> I wanted to touch the driver to see how warm they was, and I just touch the motor Y cable and it moves !!! The printer jump steps !!!
> 
> My wires are with dupont, and it seems not to connect well on the board !!
> What a mess ....
> 
> I don't how to improve connnection ...

Try re-crimping them I guess? Not sure what to tell you I have mine connected with Duponts just fine.  This isn't really a bug and the issue tracker isn't the place to voice your frustration in your efforts to get your hardware set up correctly.

davidmarli, 2020-04-24:
Yes? I know, but  I buy that card thinking it was a good board.
But reading the web, I found many users with pbs ...
What does "recrimping" mean ? Enlgish is not my forst langage

davidmarli, 2020-04-24:
Admin can delete the issue...

closed 2020-04-24 by davidmarli