bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #135: Double carriage and stallguard

abelsata, 2020-04-28:
Hi i am having trouble with dual carriage using tmc2130
When I try to activate dual carriage stallguard mode does not work on x.
If someone with an idex or who knows what may be happening would guide me, I would appreciate it.
my files are:

[Configuration .txt](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/files/4548733/Configuration.txt)
[Configuration_adv.txt](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/files/4548734/Configuration_adv.txt)

oldman4U, 2020-05-04:
Good morning.

Dual carriage on which axis?

abelsata, 2020-05-09:
Double carriage on x axis

oldman4U, 2020-05-11:
Sorry, but i can not help you. I hope someone else has a similar setup and can help.

onichan88, 2020-07-04:
Hello Abelsata, as I told you in the telegram group, you have to link the stallguard pin of the second extruser with the endstop signal of X_MAX with the cable that comes with the driver.

To not use of the cable, you can config the pins file of skr pro in the firmware of the skr pro v1.1 but I have to prove because I haven’t set that file OK.

But with the cable it works.

You can use M914 to see if the endstop its open and not trigered to see if it works. If it isn’t connected correctly, you will see that endstop as triggered.

I will post again when I get it working only modifying the pins file.

The problem is very similar to the problem with delta printer and stallguard.

oldman4U, 2020-08-26:
@abelsata 

Could you please close this ticket in case you do not need it anymore. Thank you

closed 2020-09-13 by abelsata