bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #137: SKR PRO 1.1 - Can't flash the firmware.bin with SD Card

SevCr, 2020-05-09:
Hi. 
Just purchased the above card with TFT35 and TMC2208 drivers. 
Screen says printer not connected. Downloaded new firmware for screen with new icons, which has updated ok, but it is not talking to board. 
Downloaded latest Marlin Nightly Build for the board. Compiled in VSC, copied to sd card, but nothing is happening. I put the card in, power up, screen still "not connected", can not connect SKR board to laptop via USB either. Installed USB drivers from SKR Github.
Still nothing. Can someone please give me a hand with this.
Thanks

oldman4U, 2020-05-09:
Hello SevCR.

Could you please take a picture which shows the connections you made between the SKR and the TFT.

Please also describe how you try to update the firmware and maybe you would like to attach the firmware here to your ticket.

I have a spare board where I can try things out, so I can check for you.

SevCr, 2020-05-10:
Hello oldman4U and thank you for response.

I have connected the screen following the diagram from github, this is quick sketch:
[image]

Regarding the flashing it goes like that:
1. Downloaded latest Nightly build from marlin 2.0 bugfix
2. Compiled in VSC with SKR Pro 1.1 build enviroment
3. Copy firmware.bin to SD Card
4. With SKR off plug in SDcard 
5. Power up (tried both USB and 24V dc)
6. SKR comes on, screen comes on (note, I have read somewhere that an LED shoud be flashing - I have plenty of leds on, but none are flashing)
7. TFT displaying "No printer attached"
USB connection to PC not working - cant see any devices, Pronterface cant see any COM ports active. 
I have installed USB drivers from Github:
STM_VCOM_1.3.1.rar
I have tried that many times, different USB cables, ports, with steppers plugged in and without, with steppers drivers plugged in and without. 
[firmware.bin.txt](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/files/4605661/firmware.bin.txt)
Added .txt as I couldnt attached the file here, when flashing it was just firmware.bin
I have this board for a couple of days and was unsuccesful in getting it to work. 
I you can help I will be most grateful. Thanks 
Sev

SevCr, 2020-05-10:
Pictures of card and screen:

[image]
[image]

oldman4U, 2020-05-10:
Hi.

Thank you for the pictures and the file.

I renamed the firmware.bin.txt file from you to firmware.bin, copied it on my SDcard and put it into my board. Once I powered it up, the firmware led started to flash green for about 10 sec. and stopped then. This means that the firmware has been loaded and I got it confirmed, because me TFT could not connect to the board any more (because I use 115200 Baudrate). So I changed the baud rate on the TFT to the 250000 shown on your fist picture and voila, connection established.

This means I can confirm that the firmware you sent to me can be loaded to an SKR Pro v1.1 and that a TFT display using 250000 baudrate can be used.

Before we go further we have to make sure that the firmware update all works on your machine. For this, no display and no USB connection has to be available, so please disconnect the display the USB cable and the two red cables currently connected to Hot Bed. Instead connect one + and - from your 12V or 24V power supply to the + and - labeled Power.

Take the firmware.bin file you attached to this ticket and copy it onto your SDcard. Put it into the SKR and try again to update the firmware.

Attached you can find a screenshot showing the LED which should flash during update.

Btw, are you using the SDcard which came with the SKR or another card or have you reformatted the card?

I would also recommend to change the two USB jumpers back to USB port, because USB drive doe snot work at the moment.

Please let me know
[image]

SevCr, 2020-05-10:
Hi. 
Thanks a lot for comprehensive instructions. 

I have tried that

> please disconnect the display the USB cable and the two red cables currently connected to Hot Bed. Instead connect one + and - from your 12V or 24V power supply to the + and - labeled Power.

Ive tried 2 different sd cards (one that came with board too), also tried formatting these. 
I left it for about 20-30 seconds each time, but when I checked sd card it was the same file there. Shouldnt there be renamed file "FIRMWARE"?

Also the led you mentioned does not flash at all. For all the times I tried it never flashed.

I put the jumpers back to USB as you recommended, although those were like that for many trials I had before, and just changed them recently.

Does it mean the board is faulty?

Thanks for your help.

oldman4U, 2020-05-10:
Once the firmware is flashed, the file firmware.bin is renamed too firmware.CUR, but the led must flash during upload. The funny thing is, that you can rename the .CUR to bin and you can "use" it again.

For me this sounds like a faulty board.

;-(

SevCr, 2020-05-10:
Thanks a lot for your assistance. I had feeling someting is wrong with board. 

> Once the firmware is flashed, the file firmware.bin is renamed too firmware.CUR
That is what I thought.

Anyway, I will get it exchanged and start again. 

Once again thanks a lot!

oldman4U, 2020-05-10:
Hope you have more luck with the next board.

BunniKiller, 2020-05-12:
what worked for me was ...  ( SKR Pro V1.1 with a the TFT35 E3 V3 video) I had to download the version
BIGTREETECH_SKR_PRO_V1.1_Master.zip and extract it. Once I extracted, I opened folder " FIRMWARE", then opened SKRPROV1.1_MARLIN_V2.0_12864_16div-usually folder. Inside that folder is the FIRMWARE.bin file...  copy that to a TF ( SD Micro) card and insert it into the SKR PRO board and turn on machine, let it boot and activate the file. The TFT35 should boot and allow it a few seconds for the "Printer not found" message to go away... everything works on my unit except for the stepper motors...  but Im not done with tweaking the TMC5160 drivers.  Its really odd to watch the printer go thru its paces ( watch the timer and percentages count down) and not have steppers working. Basically like having the steppers unplugged :)

BunniKiller, 2020-05-12:
on a side note, once you have the issue above solved, you can use the USB port on the TFT35 to transfer files to be printed...

oldman4U, 2020-05-12:
Hi BunniKiller - again;-)

The problem is, that even without the TFT or anything else connected to the board, the firmware SevCR sent to me and which works fine with my SKR, does not update his board at all. The ability to connect the display or stepper motors or something else would have followed, once the board accept the firmware update.

oldman4U, 2020-05-26:
@SevCr 

Have you got a new board in the meantime?

SevCr, 2020-05-26:
@oldman4U 
Hi. Thanks a lot for your message. Due to problems getting SKR Pro 1.1 I am returning the faulty board and replacing it with SKR GTR 1.0 

Would you have any recommendations as to the firmware or any particular settings for this board? I am upgrading Tronxy S5Pro (Core XY) and would like in the future to fabricate and install tool changer for 3 or 4 heads., but for now I want to get the printer to run as is. 

New board just arrived and I have so far connected it to PC and TFT and it connects through Prontaface and the screen is connected to board as well, so it seems this one is working. Havent yet tried uploading new firmware, as I suppose I need to rebuild it to different board.

While searching internet for answers for my previous issue with the SKR Pro I have come acrross some info regarding running Klipper on SKR. Would that be good idea? I have read that Klipper is good for high performance fast printing. Do you have any experience with that? 

Any advice will be much appreciate. Thanks a lot once again for your involvement.

oldman4U, 2020-05-26:
Hi SevCr.

Good to hear you got a new board. Unfortunately I have no experience with the GTR, nor have I ever tried Klipper. I am just a beginner with some knowledge related to the hardware I own, and because I got so much help from other users here, I try to give a little bit back. But because your issue is "solved" you may want to close this ticket and start new tickets for the GTR. I am sure there are others which can help you;-)

Happy printing

closed 2020-05-31 by SevCr