bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #139: HOT BED out terminals backwards

BunniKiller, 2020-05-12:
Just a heads up, the terminals ( screws) for the HOT BED output are marked opposite of the polarity that is actually being produced. I caught this by suprize.. I installed a relay for a 120VAC bed heater and the input on the relay required a DC volt input ( polarity sensitive), the relay wouldnt kick in even tho the relay led on the board showed it was on... I pulled out the volt meter to see how much voltage was being supplied to the HOT BED out terminals and found it to read -24v a.k.a. reversed polarity. So I swapped the leads on the screws and wala, bed heater relay is all happy and working now...  so, the boards polarity marking is wrong... it should read " - hot bed + "

oldman4U, 2020-05-12:
Hello BunniKiller.

This topic has been discovered right after the board has been released: https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/issues/57

It would be great, if you could close your ticket.

Thank you

BunniKiller, 2020-05-12:
Im new at this... how do I close it?

BunniKiller, 2020-05-12:
never mind i found the answer

closed 2020-05-12 by BunniKiller