bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #140: E0 don't work with all other axis working

George0023, 2020-05-12:
So, I’m trying to set up my new custom 3d printer with skr 1.1 pro, I got 3 axes working but for some reason, I can't get E0 working (I did disable cold extrusion). I tested the motors and drivers and they all work fine. So I thought maybe there is something wrong with the board, so I swapped everything to E1 and it still doesn't work. Please help.

oldman4U, 2020-05-13:
Hi George.

Stupid question. And you assigned a stepper motor driver to E0...

Happened to me, that i assigned a thermistor to the chamber instead of the bed....

What a difference a line makes - sometimes

Georg

George0023, 2020-05-14:
the heater for the hot end heat up correctly so, and when I press the extruder button I do feel the motor energize. so that shouldn't be a problem. Also, I cant get UART with tmc2208 fysetc by working with the printer. So I end up switching to ramps 1.4, and everything works fine, probably a board problem.

closed 2020-05-14 by George0023